rwaldron/johnny-five #1468: Question: How can I pass params (Full-Scale Range, and etc.) to the ADS1115

MichaelKostin, 2018-05-08:
Hi,
I've a question. I'm using ADS1115 with class Expander on Raspberry pi 3.
Here's the code: 
```javascript
var five = require("../lib/johnny-five");
var board = new five.Board();

board.on("ready", function() {
  var virtual = new five.Board.Virtual(
    new five.Expander("ADS1115")
  );

    var sensor = new five.Sensor({
      pin: virtual.io.analogPins[0],
      board: virtual
    });

    sensor.on("data", function() {
      console.log(this.value);
    });
});
```

How can i pass additional params to the ADS1115? 
I want to pass Full-Scale Range to work with voltage between 1.2 and 4 v. ADS1115 powered by 5v.
Thank you in advance.

rwaldron, 2018-05-10:
I think the J5  `Expander` controller could accept a `gain` option, but the emitted data value would still be normalized to a 10 bit uint, to ensure interoperability. Is that going to work for you?

MichaelKostin, 2018-05-13:
@rwaldron thank you for reply. I'll check the `gain` option.

MichaelKostin, 2018-05-25:
I fixed issue after adding line below. (Didn't find better way). 
```javascript 
var virtual = new five.Board.Virtual(
    new five.Expander("ADS1115")
  );

virtual.io.REGISTER.PIN_DATA = 0xc3;
```
To change the `gain` option we have to change `PIN` and  `PIN_DATA`. ADS1115 config contains 16 bits (2 numbers). Default value for pin 0 is `0xc1`, `0x83` or in binary system:  `0b11000001`, `0b10000011`.  I found it on source code [lib/expander.js#L1793](https://github.com/rwaldron/johnny-five/blob/master/lib/expander.js#L1793). The `gain` already was 6.144V. I changed only Data Rate from 128 SPS to 250 SPS (on 128 I saw zero values randomly).
To change first part of ADS config we can do something like this:

```javascript
virtual.io.REGISTER.PIN[0]= 0xc1;

```
Here is information about settings by bits.  Based on ADS1115 [datasheet](http://www.ti.com/lit/ds/symlink/ads1115.pdf). 
To convert from hexadecimal to bynary we can use `Number(0xc1).toString(2) // 11000001`
1 100 000 1, 100 0 0 0 11

[1] OS 
[100] Input multiplexer configuration (100 : AINP = AIN0 and AINN = GND)
[000] Programmable gain amplifier configuration (000 : FSR = ±6.144 V)
[1] Device operating mode (1 : Single-shot mode or power-down state (default))
[100] Data rate (100 : 128 SPS (default))
[0] Comparator mode (0 : Traditional comparator (default))
[0] Comparator polarity (0 : Active low (default))
[0] Latching comparator (0 : Nonlatching comparator)
[11] Comparator queue and disable (11 : Disable comparator) 

Hope it can be helpfull for somebody.

closed 2018-05-28 by MichaelKostin