rwaldron/johnny-five #1469: High Frequency PWM (to avoid loud motors)

ZLima12, 2018-05-11:
I am using the `Motor` class to control four motors, and although it functions well, the PWM causes a tone to be heard from the motors (which is the same frequency as the PWM). To avoid this, most people will increase the PWM frequency out of human hearing range. I know that this can be done on Arduino by making use of the registers of the microcontroller, but is there a way to do this with Johnny Five? I am willing to stop using the `Motor` class for now if need be.

dtex, 2018-05-11:
I’ll have to do some digging, but I don’t think firmata has the ability to set the PWM frequency from the host server. I think you could totally tweak the firmata sketch that gets installed on the Arduino though.

dtex, 2018-05-17:
Hi @ZLima12 

Just following up. I did find a note that said pins 5 & 6 on the Uno run at 980Hz vs 490Hz for the other pins. Is that an option for you?

ZLima12, 2018-05-20:
@dtex unfortunately, that frequency generates an audible tone.

dtex, 2018-06-06:
Sorry for the slow response. No, I don't believe there is a way to do it from Johnny-Five (with or without the motor class), since Johnny-Five depends on firmata to control the Arduino and changing the PWM frequency is not something firmata supports.

You can, however, modify firmata to use a different frequency on your motor control pins and still use Johnny-Five as-is. Note that there are some gotchas with regard to changing the frequency of the timers but depending on what other things you are doing with the Arduino that may not be an issue.

This article was the best resource I could find: https://playground.arduino.cc/Code/PwmFrequency

I would assume you would need to add a call to ```setPwmFrequency``` in the ```setPinModeCallback``` method. 

I'm going to close this issue since it is better suited for either the https://github.com/firmata/arduino or https://github.com/firmata/configurableFirmata repo.

closed 2018-06-06 by dtex

dtex, 2018-10-09:
You can only set the PWM frequency from the C code that runs on the arduino.

Which flavor (standardFirmata, standardFirmataPlus, etc) and version of firmata are you using?