rwaldron/johnny-five #1475: SI7021 does not capture a proper temperature or update

jc78, 2018-06-08:
I'm trying to use the [Sparkfun Si7021](https://www.sparkfun.com/products/13763) temp\humidity sensor, but it never seems to register a proper temp. It's always at -52.42. Additionally, when using this in combination with a [Sharp IR sensor](https://www.sparkfun.com/products/12728), the IR sensor no longer functions. Both are plugged into the analog inputs of my Arduino Uno. If I comment out the creation of the Hygrometer in my JS code, the IR sensor works again.

```
// Create the Temp sensor
var tempSensor = new five.Multi({
  controller: "SI7021"
});
tempSensor.on("data", function() {
    console.log("Current Temp: ", tempSensor.thermometer.fahrenheit};
));

// Create an IR sensor to measure distance values
var proximity = new five.Proximity({
  controller: "GP2Y0A41SK0F",
  pin: "A0",
  freq: 20
});
proximity.on("data", function() {
    console.log("Current Distance: ", proximity.cm);
});
```

- Are two analog devices permitted at the same time? 
- Are these two incompatible with each other?

It appears as if the Si7021 is never capturing a changed temp from when it is first initialized. I may end up switching to a DHT22 with a backpack if this doesn't work out. But that seems like more work to manage and I'd like to keep things as simple as possible.

Thanks for any help.

dtex, 2018-06-08:
@jc78 This should work and it's fine to use these two devices together (one is simply an analog input and the other is i2c). Can you send a picture (or pictures) of your wiring?

jc78, 2018-06-08:
@dtex, thanks for responding. I'm currently at work, but I'll send a picture later tonight.

What I can tell you is that I tried wiring the Si7021 based off of what was in the [johnny-five docs](http://johnny-five.io/img/breadboard/multi-SI7021-uno.png) and also [Sparkfun's wiring diagram](https://cdn.sparkfun.com/assets/learn_tutorials/4/6/7/Si7021Fritzing2.png). Neither of them worked for me.

Something else that I'll check when I get home is my soldering job and wire connector. I soldered the wires directly to the board, and added a connector to the other end. I did visually check those connections and all looked good. But I didn't perform a continuity test with my multimeter, which I'll try tonight.

jc78, 2018-06-09:
Problem Solved.

This ended up being a pour solder connection on the 3.3v line. Once I cleaned it up, both the Si7021 and the IR sensor started working as expected.

@dtex, thanks for the quick response!

closed 2018-06-09 by jc78