rwaldron/johnny-five #1478: Custom Characters (useChar) not working

camer0, 2018-06-17:
I am running practically the example code:
`js
lcd = new five.LCD({ 
    pins: [2, 3, 4, 5, 6, 7]
  });
  lcd.useChar("heart");
  lcd.print(":heart:");`

And only a blank character is printed to the screen. I have tried reinstalling johnny-five, but this does not fix the issue. I tested my LCD out in the Arduino IDE with a custom character and it drew it just fine, but for some reason no character will work in johnny-five. Is this a known issue or have I completely done something wrong?

camer0, 2018-06-17:
I've narrowed it down to the createChar function, specifically the

```
this.command(this.REGISTER.SETCGRAMADDR | (address << 3));
```
line in the lcd.js file.

If I remove this line, it doesn't print the character, but I can write other characters to the screen, whereas with that line I cannot write anything to the screen until I reset and remove the `useChar()` function. 

Again, custom characters works fine for me in the regular arduino IDE, but something goes amiss for me in johnny-five. Does anyone have any idea as to why this might be happening? Any help is appreciated!

rwaldron, 2018-07-12:
Sorry for the delay in responding here. I'm not sure why this would stop working—no code has changed in `LCD`. 

@fivdi is something of an expert with `LCD`, perhaps he can help us figure this out.

fivdi, 2018-07-13:
After invoking `useChar` the cursor needs to be positioned by invoking `cursor`. The following should work:

```js
var five = require("johnny-five");
var board = new five.Board();
board.on("ready", function() {
    let lcd = new five.LCD({ pins: [2, 3, 4, 5, 6, 7]});
    lcd.useChar("heart")
    lcd.cursor(0, 0); // <=== Add this to get it to work
    lcd.print("test :heart: test");
})
```

Further details: Invoking `useChar` sets the address counter on the LCD to the CGRAM address. From this point onwards all data writes are to CGRAM. Invoking `cursor` sets the address counter on the LCD to the DDRAM address. From this point onwards all data write are to DDRAM.

To display characters on the LCD the characters need to be written to the DDRAM.

camer0, 2018-07-14:
I cannot believe the problem was that simple. Thank you so much, this seems to work!

dtex, 2018-07-15:
Thanks @fivdi !

closed 2018-07-15 by dtex