rwaldron/johnny-five #1480: Possible to use with Pico Pro Maker Kit

developer239, 2018-06-26:
Hi :)
this project looks absolutely awesome.

I have this development kit:
https://www.wandboard.org/news/pico-pi-pro-maker-kit-setup-guide/

But I haven't used it at all. Basically, I just put the parts together and then I decided to do something more fun because I don't know anything about java/kotlin and everything seemed too strange to me.

Is it possible to use the development kit with the `johnny-five framework`?

I believe it has `Pico i.MX7Dual development board`.

dtex, 2018-07-15:
Sorry that there hasn't been a response before now. I don't think it's possible to use Johnny-Five on this board.

developer239, 2018-07-15:
@dtex Thank you for your answer I ended up borrowing Arduino starter kit. 🙂

closed 2018-07-15 by dtex