rwaldron/johnny-five #1481: Find correct sensor settings

nofear87, 2018-06-28:
Can you please help to find the correct sensor settings?

I have Pololu 37D Motor with enconders. See https://www.pololu.com/product/1442 -> "Using the Encoder" for detailed information.

I have connected A Output to A6 then starting with this example:

```
var sensor = new five.Sensor("A6");
sensor.on("change", function() {
   console.log(this.value);
 });
```

But getting just curious data between 0-1000. I think I haven't the correct settings now?
Is it maybe an digital signal!? Did I need hardware interupts?

Thanks!

nofear87, 2018-06-28:
> Are the values in bunches at the far ends of the 0-1000 range, all at one end, or are the spread evenly? I would expect to see a bunch near 0, then a bunch near 1000, then a bunch near 0, and so on.
>
> [
> {
> "@type": "EmailMessage",
> "@type": "ViewAction",
> "target": "https://github.com/rwaldron/johnny-five/issues/1481#issuecomment-401129555",
> "url": "https://github.com/rwaldron/johnny-five/issues/1481#issuecomment-401129555",
> "name": "View Issue"
> },
> "description": "View this Issue on GitHub",
> "publisher": {
> "@type": "Organization",
> "name": "GitHub",
> "url": "https://github.com"
> }
> },
> {
> "@type": "MessageCard",
> "@context": "http://schema.org/extensions",
> "hideOriginalBody": "false",
> "originator": "AF6C5A86-E920-430C-9C59-A73278B5EFEB",
> "title": "Re: [rwaldron/johnny-five] Find correct sensor settings (#1481)",
> "sections": [
> {
> [... 50 lines cut ...]
> }
> ],
> "themeColor": "26292E"
> }
> ]

Thanks for fast answer. They spread evenly.

dtex, 2018-06-28:
Then I would suspect a floating ground. Can I see how the encoders are wired to your microcontroller?

I assume the motor is turning as expected.

nofear87, 2018-06-28:
@dtex I switch from external power source to mega 5v. And now the values are in bunches at the far ends! I used scaleTo(0,1) to get true and false values. But how can I determine direction and stepas without an interrupt? Thank you!

ryangueraa, 2018-07-14:
nice

closed 2019-07-16 by stale[bot]

joepuzzo, 2022-08-23:
I have made this PR for firmata.js https://github.com/firmata/firmata.js/pull/261

joepuzzo, 2022-08-29:
Seems nobody has checked my PR yet.. anyone know who maintains that package ?

joepuzzo, 2022-08-29:
In order to use the encoder stuff ( until the people that own firmata.js merge my code ) you need to run this in any new project that wants to use encoder stuff

```bash
 npm i firmata-io@npm:@joepuzzo/firmata-io
```