rwaldron/johnny-five #1482: Board Closes when script run as a service on Raspberry Pi

jc78, 2018-07-05:
I have a javascript file that I'm running from a raspberry pi (NodeJS), with an Arduino Nano slaved to it. The script runs just fine if I launch it from the command line (the script is active and my components function as expected.). But when I try to get that same script to run at the Raspi's startup using systemd, the script fails every time. The output shows that the script is starting up like normal, but fails to initialize the board.

If I had to guess, I'm assuming this is a permissions issue with accessing USB ports from a service? Or something along those lines. Has anybody ever successfully ran their script as a service? My goal is to ensure it's always running; at raspi boot up, or after a crash.

thanks,
Jason

dtex, 2018-07-11:
Hi @jc78 

Sorry no reply before now. This question might be better asked over in the https://github.com/nebrius/raspi-io repository. I found [this issue comment](https://github.com/nebrius/raspi-io/issues/93#issuecomment-301269516) from last year that might help you.

dtex, 2018-07-15:
@jc78 Was that issue comment helpful for you? If so, can I close this?

PscROS, 2018-07-25:
hi,
My project is a raspberry pi3B+ (raspian latest) with 3 arduino (mega2560) boards connected by USB/serial. The program : nodejs 8.x, johnny-five and socket-io. At the begining the program load the configuration for each component of each board, and init it.
I have the same issue, i don't use raspi-io. If i launch the program from comand line by service or not, it run well.  But if i try to startup on boot the same program, it stop when J5 component init but it stop never at the same component. I don't see any error in the logs.
I don't no why but if my program do not init J5 component everything is ok.
My goal : find  a way to start J5 program at startup on raspi.
thanks

dtex, 2018-07-25:
Did you check out that issue comment mentioned above? Later n the same thread [someone said](https://github.com/nebrius/raspi-io/issues/93#issuecomment-307961690) that disabling the repl solved the problem for them.

jc78, 2018-07-25:
Sorry, I haven’t had a chance to check this out yet. I recently moved and haven’t unboxed everything including this project.

I’ll close the issue for now as the suggestion looks promising. I can always reopen this issue or create a new one if I’m still running into this problem later on.

closed 2018-07-25 by jc78

PscROS, 2018-07-25:
thank you for your quick answer.
I confirm disabling the repl solved the problem.
thank you verymuch.