rwaldron/johnny-five #1483: Device or Firmware Error A timeout occurred while connecting to the Board

ulfahputribisba, 2018-07-06:
Hi, I'm new in Arduino and Johnny-Five. I have problems during my final assignment. I'm using Arduino Uno R3 ATMega328p on Windows 10. Here's my code : 
`var five = require('johnny-five');
 
five.Board().on('ready', function() {
  var temperature = new five.Thermometer({
    controller: 'DHT11',
    pin: 13
  });
 
  temperature.on('data', function() {
    console.log(this.celsius + '°C', this.fahrenheit + '°F');
  });
});`

And here's the message that I've got : 
`1530883460513 Available COM3,COM5
1530883460776 Connected COM3
1530883470778 Device or Firmware Error A timeout occurred while connecting to the Board.

Please check that you've properly flashed the board with the correct firmware.
See: https://github.com/rwaldron/johnny-five/wiki/Getting-Started#trouble-shooting

If connecting to a Leonardo or Leonardo clone, press the 'Reset' button on the board, wait approximately 11 seconds for complete reset, then run your program again.
events.js:188
      throw err;
      ^

Error: Unhandled "error" event. ([object Object])
    at Board.emit (events.js:186:19)
    at Board.log (C:\Users\owner\nodejsjohnny\node_modules\johnny-five\lib\board.js:648:8)
    at Board.(anonymous function) [as error] (C:\Users\owner\nodejsjohnny\node_modules\johnny-five\lib\board.js:659:14)
    at Board.<anonymous> (C:\Users\owner\nodejsjohnny\node_modules\johnny-five\lib\board.js:395:14)
    at ontimeout (timers.js:498:11)
    at tryOnTimeout (timers.js:323:5)
    at Timer.listOnTimeout (timers.js:290:5)`

Hope you can help me to find the solutions.
Thanks a lot :)

dtex, 2018-07-06:
This error can appear when the firmata firmware is not installed on the board. Here are some [instructions for installing firmata](https://github.com/rwaldron/johnny-five/wiki/Getting-Started#trouble-shooting)

ulfahputribisba, 2018-07-06:
I've installed the firmware and uploaded it to the board, and it's just the same error

dtex, 2018-07-06:
Excellent! Almost there...

For thermometer (or any other analog sensor), you will need to use an Analog input pin "A0" - "A5"

dtex, 2018-07-06:
Oh wait! Doesn't the DHT11 need a backpack?

ulfahputribisba, 2018-07-06:
A backpack? What do you mean "backpack"? Can you tell me more please?

dtex, 2018-07-11:
@ulfahputribisba Was that a show stopper?

ulfahputribisba, 2018-07-11:
Hi, sorry. The Arduino Nano just come to my hand.  So, I've wired the DHT11 and Arduino Nano to the board.  When I upload this code http://johnny-five.io/examples/multi-DHT11_I2C_NANO_BACKPACK/ using ConfigurableFirmata firmware, it showed like this : 

> 1531323483921 Connected Com5
1531323489094 Repl Initialized
>> Thermometer
  celsius           :  0
--------------------------------------
Hygrometer
  relative humidity :  0
--------------------------------------

Should I upload this firmware https://github.com/rwaldron/johnny-five/blob/master/firmwares/dht_i2c_nano_backpack.ino to the board and run the johnny five code?

ulfahputribisba, 2018-07-11:
Hi, sorry. The Arduino Nano just come to my hand. So, I've wired the DHT11 and Arduino Nano to the board. When I upload this code http://johnny-five.io/examples/multi-DHT11_I2C_NANO_BACKPACK/ using ConfigurableFirmata firmware, it showed like this : 

1531323483921 Connected Com5
1531323489094 Repl Initialized
>> Thermometer
  celsius           :  0
--------------------------------------
Hygrometer
  relative humidity :  0
--------------------------------------

When I upload this firmware (https://github.com/rwaldron/johnny-five/blob/master/firmwares/dht_i2c_nano_backpack.ino) to the board and run the johnnyfive, it showed error.  

What should I do?

ulfahputribisba, 2018-07-11:
Hi, sorry. The Arduino Nano just come to my hand. So, I've wired the DHT11 and Arduino Nano to the board. When I upload this code http://johnny-five.io/examples/multi-DHT11_I2C_NANO_BACKPACK/ using ConfigurableFirmata firmware, it showed like this :

1531323483921 Connected Com5
1531323489094 Repl Initialized
Thermometer
celsius : 0
Hygrometer
relative humidity : 0

When I upload this firmware (https://github.com/rwaldron/johnny-five/blob/master/firmwares/dht_i2c_nano_backpack.ino) to the board and run the johnnyfive, it showed error.

What should I do?

dtex, 2018-07-11:
Your Uno should have StandardFirmataPlus installed and your Nano should have dht_i2c_nano_backpack.ino installed. Is that what you have?

If so, what error are you getting?

ulfahputribisba, 2018-07-12:
@dtex can you show me the example of multi dht11 and the result? I'm really confused

ulfahputribisba, 2018-07-12:
I've uploaded dht_i2c_nano_backpack.ino to my Nano and StandardFirmataPlus to my Uno, then I upload this .js code ( http://johnny-five.io/examples/multi-DHT11_I2C_NANO_BACKPACK/ ) through Windows Command Line and I got this : 

1531417744190 Connected Com5
1531417745868 Repl Initialized
Thermometer
  celsius           :  0

Hygrometer
  relative humidity :  0

How to make the DHT11 sensor read the temperature and humidity?

rwaldron, 2018-07-12:
Let's do a step by step check:

1. Connect your Nano to your computer via USB & upload dht_i2c_nano_backpack.ino via Arduino IDE.
2. Connect your UNO to your computer via USB & upload StandardFirmataPlus via Arduino IDE. 
3. Create the circuit as shown in the illustration here: http://johnny-five.io/examples/multi-DHT11_I2C_NANO_BACKPACK/
4. Make a file called "dht11.js" and put the _exact_ following code:

```js
var five = require('johnny-five');
var board = new five.Board({port : "COM5"})

board.on("ready", function() {
  var multi = new five.Multi({
    controller: "DHT11_I2C_NANO_BACKPACK"
  });

  multi.on("change", function() {
    console.log("Thermometer");
    console.log("  celsius           : ", this.thermometer.celsius);
    console.log("  fahrenheit        : ", this.thermometer.fahrenheit);
    console.log("  kelvin            : ", this.thermometer.kelvin);
    console.log("--------------------------------------");

    console.log("Hygrometer");
    console.log("  relative humidity : ", this.hygrometer.relativeHumidity);
    console.log("--------------------------------------");
  });
});
```
5. Run this from the command line: `node dht11.js`
6. If that runs correctly, replace the contents of `dht11.js` with the _exact_ following code:

```js
var five = require('johnny-five');
var board = new five.Board({port : "COM5"})

board.on("ready", function() {
  var thermometer = new five.Thermometer({
    controller: "DHT11_I2C_NANO_BACKPACK"
  });

  thermometer.on("change", function() {
    console.log("Thermometer");
    console.log("  celsius           : ", this.celsius);
    console.log("  fahrenheit        : ", this.fahrenheit);
    console.log("  kelvin            : ", this.kelvin);
    console.log("--------------------------------------");
  });
});
```

Let us know how each step goes

ulfahputribisba, 2018-07-12:
Step 1. Checked.
Step 2. Checked.
Step 3. Checked.
Step 4 and 5. The result is as below : 
`1531420164992 Connected COM5
1531420166674 Repl Initialized
Thermometer
  celsius           :  0
  fahrenheit        :  32
  kelvin            :  273.15
--------------------------------------
Hygrometer
  relative humidity :  0
--------------------------------------`
and the result just show once, not continuously.

Step 6. The result is as below : 
`1531420336415 Connected COM5
1531420341487 Repl Initialized
Thermometer
  celsius           :  0
  fahrenheit        :  32
  kelvin            :  273.15
--------------------------------------`
and the result just show once, not continuously.

Can this DHT11 I2C backpack shows the temperature and the humidity continuously like LM35?

ulfahputribisba, 2018-07-12:
Step 1. Checked.
Step 2. Checked.
Step 3. Checked.
Step 4 and 5. The result is as below :

`1531420164992 Connected COM5
1531420166674 Repl Initialized
>>Thermometer
celsius : 0
fahrenheit : 32
kelvin : 273.15

Hygrometer
relative humidity : 0
--------------------------------------`

and the result just show once, not continuously.

Step 6. The result is as below : 

`1531420336415 Connected COM5
1531420341487 Repl Initialized
>>Thermometer
  celsius           :  0
  fahrenheit        :  32
  kelvin            :  273.15
--------------------------------------`

and the result just show once, not continuously.

Can this DHT11 I2C backpack shows the temperature and the humidity continuously like LM35?

ulfahputribisba, 2018-07-12:
Step 1, checked.
Step 2, checked.
Step 3, checked.
Step 4 and 5, the result is as below : 

`1531420164992 Connected COM5
1531420166674 Repl Initialized
  Thermometer
  celsius           :  0
  fahrenheit        :  32
  kelvin            :  273.15
--------------------------------------
Hygrometer
  relative humidity :  0
--------------------------------------`

and it just show once, not continously.

Step 6, the result is as below : 

`1531420336415 Connected COM5
1531420341487 Repl Initialized
>> Thermometer
  celsius           :  0
  fahrenheit        :  32
  kelvin            :  273.15
--------------------------------------`

and it just show once, not continuously.

Can this DHT11 I2C Backpack show the temperature and humidity continuously like LM35?

ulfahputribisba, 2018-07-12:
Step 1, checked.
Step 2, checked.
Step 3, checked.
Step 4 and 5, the result is as below : 

`1531420164992 Connected COM5
1531420166674 Repl Initialized
  Thermometer
  celsius           :  0
  fahrenheit        :  32
  kelvin            :  273.15

Hygrometer
  relative humidity :  0
--------------------------------------`

and it just show once, not continously.

Step 6, the result is as below : 

`1531420336415 Connected COM5
1531420341487 Repl Initialized
 Thermometer
  celsius           :  0
  fahrenheit        :  32
  kelvin            :  273.15
--------------------------------------`

and it just show once, not continuously.

Can this DHT11 I2C Backpack show the temperature and humidity continuously like LM35?

dtex, 2018-07-15:
Just stumbled across open issue #1273 where others are having the same problem.

closed 2019-07-17 by stale[bot]