rwaldron/johnny-five #1487: 3.5" TFT LCD

mythox, 2018-07-20:
Hi, I'm very new to arduino and johnny-five, just a quick question, does Johnny-five support any 3.5" or above TFT LCD touch screen?  under the LCD category of official webpage, seems the answer is no?  thankyou

dtex, 2018-07-23:
HI @mythox 

Sorry there was no response before now. Do you have a particular display in mind?

mythox, 2018-07-24:
Hi @dtex 

no worry, it is not an issue anyway.  

and yes, here is the 3.5" TFT with shield: [https://www.banggood.com/3_5-Inch-TFT-Color-Screen-Module-320-X-480-Support-Arduino-UNO-Mega2560-p-1022298.html?cur_warehouse=CN](url)

thankyou

closed 2019-07-26 by stale[bot]

DOEHOONLEE, 2023-05-02:
@mythox , did you find a way to work it out? I recently bought this 2.4" TFT LCD and have been trying to use johnny-five to control it, but well, been failing. Any luck?