rwaldron/johnny-five #1489: Broken links in official docs in the website

mjguru, 2018-07-28:
Specifically, these are the pages from wiki which have both plural and singular versions and link from one to other (usually under the heading See Also). I am listing the links of ones I found to be broken:

1. [Boards](http://johnny-five.io/api/boards/). The link to Board at 'See also: Board.'
2. [ESC](http://johnny-five.io/api/esc/). The link to ESCs at 'If you are interested in working with multiple ESCs, see the ESCs page.'
3. [ESCs](http://johnny-five.io/api/escs/). The link to ESC at 'For working with a single ESC check out the ESC class.'
4. [Leds](http://johnny-five.io/api/leds/). The link to Led at 'For working with a single Led check out the Led class.'
5. [Motors](http://johnny-five.io/api/motors/). The link to Motor at 'For working with a single motor check out the Motor class.'
6. [Relays](http://johnny-five.io/api/relays/). The link to Relay at 'For working with a single relay check out the Relay class.'
7. [Servos](http://johnny-five.io/api/servos/). The link to Servo at 'For working with a single servo check out the Servo class.'

I think the issue underlying all these is that in the plural linked page, link to the singular version is appended to the url instead of replacing the plural version.

mjguru, 2018-07-28:
How are the docs ported from wiki to the website. I mean I am down to help if needed.

dtex, 2018-07-28:
@mjguru [This page](https://github.com/johnny-five-io/johnny-five.io) documents the process, but if you just want to edit the wiki on this repo then someone can deploy the changes to the website for you.

mjguru, 2018-07-29:
I investigated a little. It seems that any link from a main wiki page to other wiki page is broken. The reason is because we are using relative urls. So, lets say that in Motors page we are linking to Motor. We will have a relative url with href attribute 'motor'. Here's the deal: On Github Wiki, relative urls replace the word after wiki so motors becomes motor on clicking. While, in the website, the relative url gets appended to motors. So, the url becomes /motors/motor. I think instead of changing the wiki, we should make a rule or something in deployment to fix this. Maybe a script could help with relative url as ../motor say instead of motor directly.

mjguru, 2018-07-29:
Here's the deal, we have absolute links in some places while relative in others. The ones that work are those that have absolute links in wiki markdown pages. Also, there is an issue of capitalization. Like in wiki, if the link is to say Relay, then it works fine even though the actual link is relay. So, the links in wiki must also be relay and not Relay.

mjguru, 2018-07-29:
I am gonna go ahead and try changing the wiki.

mjguru, 2018-07-29:
Okay, I made relevant changes in as many wiki pages as I could. I guess we could redeploy the website? How do we do that? I am having trouble understanding the procedure in the Readme.

reconbot, 2018-08-08:
Let me know if it looks in good shape and then we can close this issue.

mjguru, 2018-08-18:
Alas no! There are still broken links for example the link from esc to escs page is broken. I guess I could take a look again.

rwaldron, 2018-08-19:
This has been fixed

- http://johnny-five.io/api/board/
- http://johnny-five.io/api/boards/
- http://johnny-five.io/api/esc/
- http://johnny-five.io/api/escs/
- http://johnny-five.io/api/led/
- http://johnny-five.io/api/leds/
- http://johnny-five.io/api/motor/
- http://johnny-five.io/api/motors/
- http://johnny-five.io/api/relay/
- http://johnny-five.io/api/relays/
- http://johnny-five.io/api/servo/
- http://johnny-five.io/api/servos/

Once someone has a chance to verify that the links in all of these are consistent and correct, we will close this issue.

reconbot, 2019-08-15:
No stalebot

reconbot, 2020-08-09:
@dtex this issue, why not?

dtex, 2020-08-09:
I'll just drop them here. Rick fixed a bunch of them, but here's what's left.

| Error | URL | Appears On |  |
|--------------------|------------------------------------------------------------|--------------------------------------------------------------------------------|------|
|404                 |http://johnny-five.io/img/j5-campaign-small-2x.gif          |http://johnny-five.io                                                           |FIELD5|
|404                                            |http://johnny-five.io/img/j5-campaign.gif                                                           |http://johnny-five.io                                                           |      |
|404                                            |http://johnny-five.io/api/Requested-Features                                                        |http://johnny-five.io/api/                                                      |      |
|404                                            |https://software.intel.com/en-us/node/672326                                                        |2 instances over more than one page                                             |      |
|404                                            |https://github.com/linino/linino_distro                                                             |http://johnny-five.io/platform-support/                                         |      |
|404                                            |http://www.pcduino.com/wiki/index.php?title=Book                                                    |http://johnny-five.io/platform-support/                                         |      |
|404                                            |http://www.mediatek.com/en/products/connectivity/gps/mt3339                                         |http://johnny-five.io/news/johnny-five-gps-class/                               |      |
|404                                            |http://johnny-five.io/examples/keypad-MPR121QR2                                                     |2 instances over more than one page                                             |      |
|404                                            |http://www.anz.ru/files/mediatek/MT7620_Datasheet.pdf                                               |http://johnny-five.io/news/tessel-2-chipkit-uno32-sparkfun-edison-arduino-block/|      |
|404                                            |http://www.amazon.com/MS5611-High-resolution-Atmospheric-Pressure-Module/dp/B00F4P6LKE?utm_source=j5|5 instances over more than one page                                             |      |
|404                                            |http://tiny-circuits.com/products/tinyduino                                                         |http://johnny-five.io/api/board/                                                |      |
|404                                            |http://www.mediatek.com/en/products/connectivity/gps/mt3339?utm_source=j5                           |http://johnny-five.io/api/gps/                                                  |      |
|404                                            |https://github.com/rwaldron/johnny-five/blob/master/docs/led-rgb-PCA9685.md                         |http://johnny-five.io/api/led.rgb/                                              |      |
|404                                            |http://osepp.com/products/sensors-arduino-compatible/osepp-ir-proximity-sensor-module               |http://johnny-five.io/api/motion/                                               |      |
|404                                            |https://github.com/rwaldron/johnny-five/releases/tag/v0.9.12                                        |3 instances over more than one page                                             |      |
|404                                            |https://www.servocity.com/html/hsr-1425cr__continuous_rotatio.html                                  |http://johnny-five.io/api/servo/                                                |      |
|404                                            |https://dl.dropboxusercontent.com/u/3531958/external-servo-power.png                                |http://johnny-five.io/api/servo/                                                |      |
|404                                            |http://www.maximintegrated.com/en/products/analog/sensors-and-sensor-interface/DS18B20.html         |http://johnny-five.io/api/thermometer/                                          |      |
|404                                            |http://johnny-five.io/api/collection                                                                |4 instances over more than one page                                             |      |
|404                                            |http://www.seeedstudio.com/wiki/images/0/03/Serial_LEC_RGB_Backlight_Lcd.jpg                        |5 instances over more than one page                                             |      |
|404                                            |http://johnny-five.io/api/orientation                                                               |http://johnny-five.io/examples/imu-bno055-orientation/                          |      |
|404                                            |http://www.maximintegrated.com/en/products/analog/sensors-and-sensor-interface/DS18S20.html         |2 instances over more than one page                                             |      |
[... cut ...]
|Timeout                                   |http://www.nkcelectronics.com/freeduino-arduino-motor-control-shield-kit.html                       |http://johnny-five.io/api/motor/                                                |      |
|Timeout                                   |http://start.tessel.io/modules/climate                                                              |3 instances over more than one page                                             |      |
|302 loop                      |http://www.atmel.com/Images/Atmel-42181-SAM-D21_Datasheet.pdf                                       |http://johnny-five.io/news/tessel-2-chipkit-uno32-sparkfun-edison-arduino-block/|      |

closed 2022-04-17 by stale[bot]