rwaldron/johnny-five #1492: original keypad of arduino

kukulako, 2018-08-17:
Greetings, why do not I see the original keypad of arduino? to program it, I'm talking about the one they always use in YouTube tutorials, I think it's called keypad 4 x 4 Matrix Array 16 Key Membrane?

rwaldron, 2018-08-19:
Only because no one has had a chance to write a Controller object for it.

dtex, 2018-08-19:
@kukulamo Do you have a link to the device for sale or documentation (or code from a different framework or language). That will help whomever decides to take this on.

dtex, 2018-11-07:
> And if we need to write the controller, I can help.

Thanks for linking the specs. It would be awesome if you'd like to take this on.

closed 2019-08-15 by dtex

AlexKutschera, 2020-02-05:
Was the required controller added and if so how's it called?