rwaldron/johnny-five #1497: GPS component event refactoring [PROPOSAL]

ajfisher, 2018-08-28:
On reviewing the current GPS component there is some inconsistency that may cause confusion around the events that get fired. Note that this is purely observational from reviewing the code for a different task so isn't born from usage.

* `data` is emitted whenever there is new data - currently undocumented. See #1496 for fixes
* `change` is emitted whenever there is a positional change
* `navigation` is emitted whenever there is a change to heading / speed

For clarity, I'd like to propose the following event changes:

* `data` stays as it is
* `change-position` is emitted for a positional change (same as current `change` event) firing on lat / lng / alt changes
* `change-navigation` is emitted for a navigational change (same as current `navigation` event) firing on speed / course changes
* `change` is emitted for any positional or navigational change.

Thus, current `change` event would stay backwards compatible as the same data would be there but then it becomes more obvious under what changes would drive the other two events.

dtex, 2018-08-29:
I agree with most of this, diverge on one point and have a few more additions:

I agree that change should be emitted for any positional or navigational change and that we should add a tolerance on the position (with a default of 5 meters). Navigation should not have a tolerance since small navigation changes can have a big impact.

I don't think we need to prepend change- to position and navigation and that those events should just be emitted for every read of that data from the GPS (they are throttled by default).

The unknown event should be blown away and we should return an error as the first callback arg to all the events.

Fix the sentence/message flub in the docs. Not sure which is the right way to go since not all GPS's use NMEA sentences. I'll figure it out.

Wrap "sentence" string in an object for consistency.

operations events should pass a satellite data object, not a raw sentence.

ntatko, 2018-12-03:
Please make the tolerance on position exposed. Those relatively little jumps keep a Kalman filter *super* accurate. The changes to the change event sound good, though.

closed 2021-01-23 by stale[bot]