rwaldron/johnny-five #1498: Autoreconnect EtherPortClient

jerosoler, 2018-08-29:
Is it possible to auto-connect EtherPortClient ??

I have been doing tests with **Johnny** and when disconnecting and connecting does not reconnect.
### Johnny-Five (Not Restart) :-1:
Code: 
```javascript
var five = require('johnny-five');
var Firmata = require("firmata");
var EtherPortClient = require('etherport-client').EtherPortClient;

var boards = new five.Boards([
  {
      id: "A",
      io: new Firmata(new EtherPortClient({
            host: '192.168.1.20',
            port: '3030'
        }) ),
        repl: true,
        timeout: 1e5
  },
  {
    id: "B",
    io: new Firmata('/dev/ttyACM0'),
    repl: true
  }
])
boards.on("ready", function() {
  var led = new five.Led({
    pin: 13,
    board: this.byId("A")
  });

  this.repl.inject({
    on: function() {
     led.on();
   },
   off: function() {
     led.off();
   }
  });

  var led2 = new five.Led({
    pin: 13,
    board: this.byId("B")
  });

  this.repl.inject({
    on2: function() {
     led2.on();
   },
   off2: function() {
     led2.off();
   }
  });

});

```

### Firmata (Restart) :+1: 
If we use only **Firmata**, it auto-connects alone.  Each independent board is connected.
Code:
```javascript
var Firmata = require("firmata");
var repl = require('repl');
var EtherPortClient = require('etherport-client').EtherPortClient;

const cmd = repl.start({
  prompt: ">> ",
  useGlobal: false
});

var board = new Firmata(new EtherPortClient({ host: '192.168.1.20', port: '3030' }));
var boardB = new Firmata('/dev/ttyACM0');

board.on("ready", function() {
  console.log("READY A");
  cmd.defineCommand('off', function () {
      board.digitalWrite(13, 0);
   });

   cmd.defineCommand('on', function () {
       board.digitalWrite(13, 1);
    });
});

boardB.on("ready", function() {
  console.log("READY B");
  cmd.defineCommand('off2', function () {
      boardB.digitalWrite(13, 0);
   });

   cmd.defineCommand('on2', function () {
       boardB.digitalWrite(13, 1);
    });
});

```

### cylon  (Restart) :+1:
I have also checked with **cylon** and it is also possible.
Code:
```javascript 
var Cylon = require('cylon');
var repl = require('repl');
var EtherPortClient = require('etherport-client').EtherPortClient;

const cmd = repl.start({
  prompt: ">> ",
  useGlobal: false
});

Cylon.robot({
  connections: {
    nodemcu: { adaptor: 'firmata', port: new EtherPortClient({ host: '192.168.1.20', port: '3030' }) },
    arduino: { adaptor: 'firmata', port: '/dev/ttyACM0' }
  },

  devices: {
    led: { driver: 'led', pin: 13, connection: "nodemcu"},
    led2: { driver: 'led', pin: 13, connection: "arduino" }
  },

  work: function(my) {

    cmd.defineCommand('off', function () {
        my.led.turnOff();
     });

     cmd.defineCommand('on', function () {
         my.led.turnOn();
      });

  }
}).start();

```

Is there any way? Can it be implemented?

I've searched the issues but they all refer to the USB port.

regards

jerosoler, 2018-08-30:
Is it a possible solution
```javascript
var five = require('johnny-five');
var Firmata = require("firmata");
var EtherPortClient = require('etherport-client').EtherPortClient;

var connection = 0;
var sp = new Firmata(new EtherPortClient({
      host: '192.168.1.20',
      port: '3030'
  }));

sp.on("connect", function() {
  if(connection > 0 ) {
      console.log("Emit Ready: "+ connection)
      connection++;
      board.emit("ready");

  } else {
    connection++;
    //AUTO CONNECT PING
    setInterval(function(){ sp.sysexCommand([0]); }, 3000);
  }

});

var board = new five.Board(
  {
      id: "A",
      io:  sp,
        repl: true,
        timeout: 1e5
  });

board.on("ready", function() {
  console.log("Ready Board ID: " + board.id);
  console.log("Connection: " + connection);

  var led = new five.Led(13);

  this.repl.inject({
    on: function() {
     led.on();
   },
   off: function() {
     led.off();
   },
   stop: function() {
     led.stop();
   },
   blink: function() {
     led.blink(500);
   }
  });
});

```

TeresaPlaz, 2018-10-01:
Mine connects automatically each time i run node. Are you trying to disconnect and reconnect with repl?

jerosoler, 2018-10-01:
Your code produces a loop

closed 2019-10-03 by stale[bot]