rwaldron/johnny-five #1499: TypeError: keyFrames.map is not a function

arindamghoshal, 2018-09-02:
Not sure what I am doing. I am getting this error while animating servos based on the code in eg/phoenix.js. I am trying to animate 4 legs. Attached is the code file. The error gets triggered while executing bot.sleep(); [line 109 in the attached file]

TypeError: keyFrames.map is not a function
    at Servo.(anonymous function) [as @@normalize] (/home/pi/projects/quadBot/node_modules/johnny-five/lib/servo.js:346:20)
    at Servos.<anonymous> (/home/pi/projects/quadBot/node_modules/johnny-five/lib/servo.js:646:46)
    at Array.map (<anonymous>)
    at Servos.(anonymous function) [as @@normalize] (/home/pi/projects/quadBot/node_modules/johnny-five/lib/servo.js:620:22)
>> 

[index.js.zip](https://github.com/rwaldron/johnny-five/files/2343521/index.js.zip)

dtex, 2018-09-02:
Hmmm, it looks like you've properly written your code to handle just four legs so the problem doesn't jump out at me. I'll have to some hardware from the lab this afternoon to run the code (and console.log a lot).

One thing does jump out at me, though it has nothing to do with your current issue... I've never successfully controlled multiple servos at a high rate of speed over I2C. I think there is some limit as to how fast those messages can be sent, and even though the PCA9685 can receive instructions for multiple servos in one message, we are only sending one at a time from Johnny-Five.

Also, there is a newer, better way to control walking robots. It's not as well documented (there is no book to go with it), but https://github.com/dtex/tharp uses an inverse kinematics layer on top of Johnny-Five to make controlling the robot much easier (and dynamic).

I guess I really only need an Arduino to run the code with little modification. I'll see if I can replicate the error.

arindamghoshal, 2018-09-02:
Thanks @dtex. I will look into https://github.com/dtex/tharp library.

dtex, 2018-09-03:
I was able to replicate. Still investigating.

dtex, 2018-09-03:
We have a collection of a collection of servos. Your code (like mine) expects the coxa, femur, tibia collections to all be passed the same value so we should be normalizing them using the Servo normalize method instead of the Servos normalize method. For some reason they are being normalized as Servos even though we have insufficient keyFrames (which is why the error is being thrown). 

I changed the ```sleep``` object to this dirty hack and it got past the error:

````js
const sleep = {
        target: bot.joints,
        duration: 500,
        cuePoints: [0, 0.5, 1.0],
        fps: 100,
        oncomplete: function() {
            bot.state = "sleep";
        },
        keyFrames: [
            [
                [null, false, { degrees: 45, easing: easeOut }],
                [null, false, { degrees: 45, easing: easeOut }],
                [null, false, { degrees: 45, easing: easeOut }]
            ],
            [
                [null, { degrees: 136, easing: easeInOut }, { degrees: 180, easing: easeInOut }],
                [null, { degrees: 136, easing: easeInOut }, { degrees: 180, easing: easeInOut }],
                [null, { degrees: 136, easing: easeInOut }, { degrees: 180, easing: easeInOut }]
            ],
            [
                [null, { degrees: 120, easing: easeInOut }, { step: 60, easing: easeInOut }],
                [null, { degrees: 120, easing: easeInOut }, { step: 60, easing: easeInOut }],
                [null, { degrees: 120, easing: easeInOut }, { step: 60, easing: easeInOut }]
            ]
        ]
    };
````

Unfortunately it just hit another error so it really needs to be fixed properly. I'm going to try some old versions of Johnny-Five and see if I can find where we stopped getting the expected behavior.

dtex, 2018-09-03:
It broke at v0.10.1 so if you roll back to v0.10.0 it should work. I'll get to work on the real fix.

dtex, 2018-09-11:
Hi @arindamghoshal 

I think I have a fix for you at [this PR](https://github.com/rwaldron/johnny-five/pull/1503). It passes unit tests, but I'd really appreciate real world testing if you have a chance.

arindamghoshal, 2018-09-11:
Thanks, @dtex. Happy to test on my bot which is still work in progress.

arindamghoshal, 2018-09-14:
I took the updated servo.js and tested the same code and received a different error in animation.js

>> /home/pi/projects/quadBot/node_modules/johnny-five/lib/animation.js:473
        calcValue = this.target[Animation.keys].reduce(function(accum, key) {
                                                ^

TypeError: Cannot read property 'reduce' of undefined
    at Animation.<anonymous> (/home/pi/projects/quadBot/node_modules/johnny-five/lib/animation.js:473:49)
    at Array.map (<anonymous>)
    at Animation.tweenedValue (/home/pi/projects/quadBot/node_modules/johnny-five/lib/animation.js:422:41)
    at Animation.loopFunction (/home/pi/projects/quadBot/node_modules/johnny-five/lib/animation.js:293:38)
    at Immediate.processQueue (/home/pi/projects/quadBot/node_modules/temporal/lib/temporal.js:195:20)
    at runCallback (timers.js:693:18)
    at tryOnImmediate (timers.js:664:5)
    at processImmediate (timers.js:646:5)
    at process.topLevelDomainCallback (domain.js:121:23)

dtex, 2018-09-15:
Ah, my tests only checked the keyframe normalization, not the rendering. I see the cause of this new error now and am working on a solution.

dtex, 2018-09-15:
Okay, it's good for another go. I've updated servo.js again. Servo animation was originally crafted to receive keyFrame values with the property ```degrees```, but was changed to expect the more generic property ```value```. I was catching that in the normalization method for a single Servo, but not a collection of Servos.

arindamghoshal, 2018-09-15:
Just ran the same code, the servos are reacting good. No more errors. My movement degrees are off, which I need work on. Thanks @dtex.

dtex, 2018-09-15:
This is great news. This issue will close when I land that PR. I want to add some tests and bump the version so it'll be a couple of days, but keep an eye out for version 1.0.1.

Best of luck with your bot. Oh, and please share video(s) when you get it up and running. Nothing makes us happier than seeing nodebots.

dtex, 2018-09-22:
The fix has landed

closed 2018-09-22 by dtex