rwaldron/johnny-five #1505: Device or Firmware Error with WeMos D1 Mini

RiLe1806, 2018-09-25:
Hey guys,
I have been trying for days now to get my Wemos D1 Mini working. I couldn't find an issue that actually solved my problem. 

The board itself is working fine. I can upload sketches without any problems. My other Wemos board has the same issue. I did upload the StandardFirmata beforehand. I also tried the StandardfirmataPlus and the StandardFirmataWifi with Etherportclient. Drivers are installed and the device-manager shows the board correctly.

But I keep getting this error:

***1537888667249 Connected COM3
1537888677253 Device or Firmware Error A timeout occurred while connecting to the Board.***

***Please check that you've properly flashed the board with the correct firmware.
See: https://github.com/rwaldron/johnny-five/wiki/Getting-Started#trouble-shooting***

***If connecting to a Leonardo or Leonardo clone, press the 'Reset' button on the board, wait approximately 11 seconds for complete reset, then run your program again.
events.js:173
    throw err; // Unhandled 'error' event
    ^***

***Error [ERR_UNHANDLED_ERROR]: Unhandled error. ([object Object])
    at Board.emit (events.js:171:17)
    at Board.EventEmitter.emit (domain.js:442:20)
    at Board.log (C:\Users\GLaDoS\Documents\Smartlock\smartlock_server\node_modules\johnny-five\lib\board.js:648:8)
    at Board.(anonymous function) [as error] (C:\Users\GLaDoS\Documents\Smartlock\smartlock_server\node_modules\johnny-five\lib\board.js:659:14)
    at Board.<anonymous> (C:\Users\GLaDoS\Documents\Smartlock\smartlock_server\node_modules\johnny-five\lib\board.js:395:14)
    at ontimeout (timers.js:425:11)
    at tryOnTimeout (timers.js:289:5)
    at listOnTimeout (timers.js:252:5)
    at Timer.processTimers (timers.js:212:10)***

#### My Code looks like this:

```javascript
const SerialPort = require("serialport");
const five = require("johnny-five");
const board = new five.Board({
  port: "COM3"
});

board.on("ready", function() {
  console.log("READY!");
  
  var led = new five.Led(2);
  led.blink(500);
});

SerialPort.list(function(err, ports) {
    console.log(ports);
}); 
```

#### The SerialPort.list() returns the following:

***[ { comName: 'COM1',
    manufacturer: '(Standardanschlusstypen)',
    serialNumber: undefined,
    pnpId: 'ACPI\\PNP0501\\1',
    locationId: undefined,
    vendorId: undefined,
    productId: undefined },
  { comName: 'COM3',
    manufacturer: 'wch.cn',
    serialNumber: '5&e658374&0&10',
    pnpId: 'USB\\VID_1A86&PID_7523\\5&E658374&0&10',
    locationId: 'Port_#0010.Hub_#0003',
    vendorId: '1A86',
    productId: '7523' } ]***

Any ideas? I really want to get this working but I'm out of ideas

Thanks!

dtex, 2018-09-25:
The Wemos D1 Mini is an ESP8266 based board so it expects to communicate over standardFirmataWifi. I did this for the first time this past weekend so I'm no expert, but here is the [article I found most helpful](https://www.instructables.com/id/ESP8266-Firmata-J5-NodeBot/) (note that a lot of the other how-to's out there were written as the support itself was being developed so many are no longer current).

The hiccups I ran into that weren't covered in this article are:

1. Getting the IP addressed that was assigned to the ESP8266 by your DHCP server was not covered. If you open the serial monitor in the Arduino IDE and turn the board on, the wifi specs will be output when the board finished booting/connecting (you may need to fiddle around withthe baud to get something you can read).
2. Your wifi network may have support for 2.4 and 5ghz networks. You must make sure that your computer and the board are both connected to the same network. The easiest way to do this is to rename the 2.4 Gig network at your wi-fi router so that it is easy to differentiate the two.

RiLe1806, 2018-09-25:
I have tried it via Wifi before and it didn't work out. But you gave me the right clue, thanks.

I changed my DHCP range and well ... it works now.

dtex, 2018-09-25:
That's great! I'll close this for now. If you have any other problems feel free to re-open.

closed 2018-09-25 by dtex