rwaldron/johnny-five #1509: Angular

leomozzer, 2018-10-05:
How to integrate the johnny-five with angular6. In my component.ts I have to import the j5 to turn on the led, but don't know how to do the import in my component

dtex, 2018-10-05:
Angular is a front-end framework correct? Can you tell us more about the stack? It's not obvious to me what role Angular is playing.

leomozzer, 2018-10-06:
Yes, it's a front-end framework. I created a component which have some buttons, the components are a simple way to create archives like html, css, and ts files, different components have different archives html, css and ts files. I created the buttons in my html file, what I know is all my functions are write in ts files, so I think that I need to import my j5 library, but don't know how to import.

MichaelKostin, 2018-11-23:
@leomozzer Looks like your Angular code will work in browser. There are no way to contact with boards (arduino, raspberry, ...) directly from browser. It make sense to use jonny-five on the backend side and contact with it using API (Websockets and so on).

leomozzer, 2018-11-24:
Thanks, i found a way to conect, i used a nodejs server wich run the johnny-five and a angular project that consumes the sever

closed 2018-11-24 by leomozzer