rwaldron/johnny-five #1513: Error: Stepper cannot exceed max steppers (6)

arroyoc, 2018-10-26:
Hello, I am able to run my function a total of 6 times. After that I get the following error:

```
Error: Stepper cannot exceed max steppers (6)
    at new Stepper (C:\Users\chris\Documents\cc_swipe\node_modules\johnny-five\lib\stepper.js:143:11)
    at app.get (C:\Users\chris\Documents\cc_swipe\index.js:13:16)
    at Layer.handle [as handle_request] (C:\Users\chris\Documents\cc_swipe\node_modules\express\lib\router\layer.js:95:5)
    at next (C:\Users\chris\Documents\cc_swipe\node_modules\express\lib\router\route.js:137:13)
    at Route.dispatch (C:\Users\chris\Documents\cc_swipe\node_modules\express\lib\router\route.js:112:3)
    at Layer.handle [as handle_request] (C:\Users\chris\Documents\cc_swipe\node_modules\express\lib\router\layer.js:95:5)
    at C:\Users\chris\Documents\cc_swipe\node_modules\express\lib\router\index.js:281:22
    at Function.process_params (C:\Users\chris\Documents\cc_swipe\node_modules\express\lib\router\index.js:335:12)
    at next (C:\Users\chris\Documents\cc_swipe\node_modules\express\lib\router\index.js:275:10)
    at expressInit (C:\Users\chris\Documents\cc_swipe\node_modules\express\lib\middleware\init.js:40:5)
```

Does anyone know how to fix this?

Here is my code that I am running:
```
const express = require('express')
const app = express()
const port = 3000
var five = require("johnny-five");
var board = new five.Board({ 
	//port: "COM1"
});

app.get('/validswipe', (req, res) => {

  //board.on("ready", function() { 
	var k = 0; 
	var stepper = new five.Stepper({ 
		type: five.Stepper.TYPE.DRIVER, 
		stepsPerRev: 200, 
		pins: [5, 2] 

	});

stepper.rpm(50).ccw().step(105, function() { 
	stepper.rpm(25).cw().step(105, function() { 
		console.log("first"); 
	}); 
	console.log("Done stepping!"); 
});

//});
  
  res.send('Hello World!')

})

app.listen(port, () => console.log(`Example app listening on port ${port}!`))
```

Other information:
My stepper motor:
https://www.amazon.com/gp/product/B074JJ6NGQ/ref=oh_aui_detailpage_o07_s00?ie=UTF8&psc=1

My stepper motor driver:
https://www.amazon.com/gp/product/B06Y5VPSFN/ref=oh_aui_detailpage_o07_s00?ie=UTF8&psc=1

My arduino is an arduino uno.

dtex, 2018-10-26:
Replying from my phone so it’s not easy to give a code example, but I’ll explain it as best I can.

You are re-instantiating the stepper motor every time I request is made for validswipe.

The stepper should be instantiated just one time in the board.on(“ready”) handler, and then just call the move inside the validswipe route.

The only tricky thing here is making sure that the board is ready in the steps of been initialized before you try to move them in your route.

dtex, 2018-10-27:
I think this will work for you. Also adds the delay requested in the other issue.

````js
const express = require('express')
const app = express()
const port = 3000
var five = require("johnny-five");

var board = new five.Board();

var stepper;
let ready = false;

board.on("ready", function() {
    var k = 0; 
    stepper = new five.Stepper({ 
        type: five.Stepper.TYPE.DRIVER, 
	stepsPerRev: 200, 
	pins: [5, 2] 
    });
    ready = true;
});

        

app.get('/validswipe', (req, res) => {
    if (ready) {
        stepper.rpm(50).ccw().step(105, function() { 
	    setTimeout(() => {
                stepper.rpm(25).cw().step(105, function() { 
		    console.log("Done stepping!"); 
	        }); 
             }, 500);
            res.send('Hello World!')
        });
    } else {
        res.send('Board not ready!');
    }
});

app.listen(port, () => console.log(`Example app listening on port ${port}!`))
````

arroyoc, 2018-10-29:
@dtex Thank you for giving me that example! It is working flawlessly and exactly how I was wanting it. I am no longer getting the error.

closed 2018-10-29 by arroyoc