rwaldron/johnny-five #1518: shiftregister.

libbie86, 2018-11-07:


libbie86, 2018-11-07:
Hi I am getting the following error when I try and create a new shiftRegister

/node_modules/johnny-five/lib/shiftregister.js:35
  this.pins.reset = typeof opts.pins.reset !== "undefined" ? opts.pins.reset : null;
                                     ^

TypeError: Cannot read property 'reset' of undefined
    at new ShiftRegister (/nodejs/easyconnect-v1/node_modules/johnny-five/lib/shiftregister.js:35:38)
    at Board.<anonymous> (/nodejs/easyconnect-v1/test.js:8:18)
    at emitNone (events.js:111:20)
    at Board.emit (events.js:208:7)
    at _combinedTickCallback (internal/process/next_tick.js:131:7)
    at process._tickDomainCallback (internal/process/next_tick.js:218:9)
    at Function.Module.runMain (module.js:695:11)
    at startup (bootstrap_node.js:191:16)
    at bootstrap_node.js:612:3

dtex, 2018-11-07:
Hi @libbie86 

Can you share your code?

libbie86, 2018-11-07:
Hi @dtex 

yeah sure.....

`var five = require("johnny-five");
var Raspi = require("raspi-io");
var board = new five.Board({io: new Raspi()});

// For use with 74HC595 chip

board.on("ready", function() {
  var register = new five.ShiftRegister({
      data: 'GPIO26',
      clock: 'GPIO20',
      latch: 'GPIO21',
      reset: 'GPIO16',
    });

  var value = 0;

  setInterval(function() {
    value = value > 0x11 ? value >> 1 : 0x88;
    register.send(value);
  }, 200);
});`

dtex, 2018-11-07:
You need to either pass the pin configuration as an array:

````js
var register = new five.ShiftRegister(['GPIO26', 'GPIO20', 'GPIO21', 'GPIO16']);
````

or as an object with a ```pins``` property like this:
````js
var register = new five.ShiftRegister({ 
  pins: {
    data: 'GPIO26',
    clock: 'GPIO20',
    latch: 'GPIO21',
    reset: 'GPIO16'
  }
});
````

libbie86, 2018-11-07:
Thanks @dtex

closed 2018-11-07 by libbie86