rwaldron/johnny-five #1520: Sensor disable function only work once

riodajs, 2018-11-09:
I have this code:
```
myBoard.on("ready", function(){
  myMic = new Sensor("A0", {"enabled": false});
  myLed = new Led(13);
  setInterval(function(){
      myMic.enable();
      console.log("Mic ON");
      myMic.on("data", (val) => {
        console.log("Analogico: " + val);
      });
      myMic.disable();
      console.log("Mic OFF");
    }
  }, 3000);
});
```
The microphone is only deactivated once, then the disable function look does not work correctly.

dtex, 2018-11-09:
At first blush it looks like state.enabled isn't managed properly in the ```enable()``` method. I won't be able to look at it until tomorrow, but maybe somebody else will see the issue before then.

dtex, 2018-11-11:
Okay, looking a little closer that may not be the problem here. 

Are you just wanting to get a reading once every three seconds?

dtex, 2018-11-11:
Some comments about your code:

Sensor params can be ```Sensor(pin)``` or ```Sensor(opts)```.  This is the correct signature for what you want to do:

````js
myMic = new Sensor({ pin: "A0", "enabled": false});
````

In your setInterval function you are adding an additional listener every three seconds. That should be added just one time, outside the interval:
````js
myMic.on("data", (val) => {
  console.log("Analogico: " + val);
});
````

But I don't think you even need the setInterval. It looks like you are trying to get a reading once every three seconds. I don't think your code will ever do that since the enable, disable and console log commands are synchronous and the sensor will never be enabled when data is received. I think this is what you want:
````js
const five = require("johnny-five");
const myBoard = new five.Board();

myBoard.on("ready", function(){
  myMic = new five.Sensor({ pin: "A0", "freq": 3000});
  myMic.on("data", (val) => {
    console.log("Analogico: " + val);
  });
});
````
All that said, there may still be a bug in enable/disable but it shouldn't be a blocker for you.

riodajs, 2018-11-16:
Thanks, but this was only an example, the idea is create a button to enable and disable the sensor.

closed 2019-11-18 by stale[bot]

Tsteinroesland, 2023-12-29:
I'm also seeing this issue with a very simple approach:

Code to reproduce:

`const { Board, Sensor } = require('johnny-five')

const board = new Board({
  port: 'COM3',
})

board.on('ready', () => {
  const moistureSensor = new Sensor({
    pin: 'A0',
    freq: 100,
    enabled: false,
  })

  moistureSensor.on('data', value => {
    console.log(value)
  })

  setTimeout(() => {
    console.log('Disable')
    moistureSensor.disable()
  }, 1000)

  setTimeout(() => {
    console.log('Enable')
    moistureSensor.enable()
  }, 2000)

  setTimeout(() => {
    console.log('Disable')
    moistureSensor.disable()
  }, 3000)
})
`

Console:

1703845514841 Connected COM3
1703845518633 Repl Initialized
>> Disable
Enable
410
410
410
410
410
410
410
410
410
Disable
410
[.... Keeps logging forever]