rwaldron/johnny-five #1536: Can not run with brand new uno board

liesislukas, 2018-11-23:
following instructions from https://github.com/rwaldron/johnny-five/wiki/Getting-Started#trouble-shooting

> Flash the arduino board via arduino --board "arduino:avr:uno" --upload ./path/to/firmware/firmware.ino. Remember to change your board according to what you use. See below on how to figure out that identifier.

This command is probably old. as well as `arduino --install-library "Firmata,Servo"` 

This is the output if i run `arduino board` (--board flag is unknown):

```
Arduino board commands.

Usage:
  arduino-cli board [command]

Examples:
  # Lists all connected boards.
  arduino-cli board list

  # Attaches a sketch to a board.
  arduino-cli board attach serial:///dev/tty/ACM0 mySketch

Available Commands:
  attach      Attaches a sketch to a board.
  details     Print details about a board.
  list        List connected boards.
  listall     List all known boards and their corresponding FQBN.

Flags:
  -h, --help   help for board

Global Flags:
      --config-file string   The custom config file (if not specified ./.cli-config.yml will be used).
      --debug                Enables debug output (super verbose, used to debug the CLI).
      --format string        The output format, can be [text|json]. (default "text")

```

My JavaScript code:
```
var five = require("johnny-five");
var board = new five.Board();

board.on("ready", function () {

  // Create a standard `led` component instance
  var led = new five.Led(13);

  // "blink" the led in 500ms
  // on-off phase periods
  led.blink(500);
});
```
Output:

```
1543013398208 Available /dev/tty.usbmodem145230  
1543013398215 Connected /dev/tty.usbmodem145230  
1543013408220 Device or Firmware Error A timeout occurred while connecting to the Board. 

Please check that you've properly flashed the board with the correct firmware.
See: https://github.com/rwaldron/johnny-five/wiki/Getting-Started#trouble-shooting

If connecting to a Leonardo or Leonardo clone, press the 'Reset' button on the board, wait approximately 11 seconds for complete reset, then run your program again.  
events.js:188
      throw err;
      ^

Error: Unhandled "error" event. ([object Object])
    at Board.emit (events.js:186:19)
    at Board.log (/Users/lukas/Desktop/arduino/firmata_1/node_modules/johnny-five/lib/board.js:648:8)
    at Board.(anonymous function) [as error] (/Users/lukas/Desktop/arduino/firmata_1/node_modules/johnny-five/lib/board.js:659:14)
    at Board.<anonymous> (/Users/lukas/Desktop/arduino/firmata_1/node_modules/johnny-five/lib/board.js:395:14)
    at ontimeout (timers.js:498:11)
    at tryOnTimeout (timers.js:323:5)
    at Timer.listOnTimeout (timers.js:290:5)
[nodemon] app crashed - waiting for file changes before starting...
```

liesislukas, 2018-11-23:
All i needed is to go to examples and upload StandardFirmata to Uno. Now it started to work.

liesislukas, 2018-11-23:
http://johnny-five.io/examples/board/ first example should have short paragraph saying what you need to do to run any sample.

dtex, 2018-11-26:
We should also update the troubleshooting section of the Getting Started wiki page.

liesislukas, 2018-11-27:
the best thing would be a short video. From stock UNO board to working for the johnny-five board. I missed it a lot. Also, I had issues with Windows, glad I have also a macos machine, so just switched to that one. So would be best to have a video of the same thing for each platform. Windows, MacOS and probably Ubuntu. First of all, it would be great material to learn from and also great content to share & find on youtube.

Caine78, 2018-11-30:
> the best thing would be a short video. From stock UNO board to working the johnny-five board. 

Yes. I don't know when we will convince them to start doing the videos. It is so much easier to follow somebody else, rather than picking it up from multiple pages and posts like this one. In a video, you can add comments and people can also discuss it. Like a post, just a bit more 21 century.

I was thinking in the past to create the website, where there will be an option to add the extra text or video, between the lessons. You will be sort of doing 101 courses, but you can always skip to the bit that you need and read the additional information about troubleshooting. Done by users and admins to find the workarounds. All in 1 place. Unfortunately, it is a way too big project for me. I was more thinking about how to layout the page, rather than fill it in. Maybe users will be able to create own menus and sort the way how they would like to view the tutorials, base on the key skills of the user. You would come there to find or share something that you did and you will contribute to the project. Almost like over here, but based on video, images and text.

I just love how the 3ds max used to have the manual on the website, back to 2017. I think that they just started messing around with it. It was almost perfect, just more videos were needed. They were using a lot of animation gif's, to show the steps. Step by step you were introduced to this world. I started using it over 20 years ago. I am still coming back to manual and I can find what I want within seconds. The old manual, not the new one. (I think they changed something, as I was just playing around with the new version a few months back. I am not 100% sure, that they actually broke it.

AtomicCons, 2019-01-25:
@dtex 
i updated https://github.com/rwaldron/johnny-five/wiki/Getting-Started#troubleshooting on using IDE to put the board in the proper state

closed 2020-01-27 by stale[bot]

uxmarinho, 2021-02-19:
> Tudo que eu precisava era ir aos exemplos e fazer o upload do StandardFirmata para o Uno. Agora começou a funcionar. Essa página wiki é enganosa.

Running StandardFirmata worked for me, Arduino Mega 2560