rwaldron/johnny-five #1537: Raspberry PI - JavaScript heap out of memory

LibbieLibbieLoo, 2018-11-29:
Hi,

I'm running johnny five on a raspberry pi controlling a couple PCA9685, everything works fine to start with but after code has run for a bit I got the following 

"FATAL ERROR: CALL_AND_RETRY_LAST Allocation failed - JavaScript heap out of memory"

I run the app again with --max_old_space_size=3000, the app run for a bit longer then I got "Killed" consoled logged.

Every know and then just before it crashes the app seems to purse for a second or 2, not sure if this is related.

LibbieLibbieLoo, 2018-11-29:
Hi Dtex,

Current its this

const five = require("johnny-five");
const raspiio = require("raspi-io");
const board = new five.Board({io: new raspiio(), repl: false, debug: false});

var flashStatus;

const numberArry = [
    "0111111A",
    "0000110A",
    "1011011A",
    "1001111A",
    "1100110A",
    "1101101A",
    "1111101A",
    "0000111A",
    "1111111A",
    "1100111A"
  ];

const blank = "0000000A";

const grn_adr = 0x40;
const red_adr = 0x42;
const amb_adr = 0x41;

function clear(colour){

    var adr;

    if(colour=="GREEN"){
        adr = grn_adr;
    }else if (colour=="RED"){
        adr = red_adr;
    }else if (colour=="AMBER"){
        adr = amb_adr;
    }

    digit_binary = blank.concat(blank);
    var digitArr = digit_binary.split("");

    for (let bit = digit_binary.length-1; bit >= 0; bit--){
        var setbit = digitArr[bit];

        if(setbit!="A"){
            var segPin = 14-bit;

            const segment = new five.Led({pin: segPin, address: adr, controller: "PCA9685"});

            if(setbit==1){
                segment.brightness(brightness);
            }else{
                segment.off();
            }     
        }
    }
}

module.exports = {

    setDigits: function (speed, brightness, colour, cb){

        var ones_binary;
        var tens_binary;
        var digit_binary;
        var adr;

        if(colour=="GREEN"){
            adr = grn_adr;
        }else if (colour=="RED"){
            adr = red_adr;
        }else if (colour=="AMBER"){
            adr = amb_adr;
        }

        ones = speed%10;
        tens = Math.floor((speed/10)%10);

        if(speed>0&&speed<100){
            ones_binary = numberArry[ones];
        }else{
            ones_binary = blank;
        }

        if(speed>9&&speed<100){
            tens_binary = numberArry[tens];
        }else{
            tens_binary = blank;
        }

        digit_binary = tens_binary.concat(ones_binary);
        var digitArr = digit_binary.split("");

        for (let bit = digit_binary.length-1; bit >= 0; bit--){
            var setbit = digitArr[bit];

            if(setbit!="A"){
                var segPin = 14-bit;

                const segment = new five.Led({pin: segPin, address: adr, controller: "PCA9685"});

                if(setbit==1){
                    segment.brightness(brightness);
                }else{
                    segment.off();
                }     
            }
        }
        cb("Digits Set");
    },

    clearDigits: function(colour){
        console.log("Clearing Digits");

        if(colour!=null){
            clear(colour);
        }else{
            clear("GREEN");
            clear("AMBER");
            clear("RED");
        }
    }
}

and setDigits is being called from the main app in a loop counting from 1 to 100 every 250ms, which works perfectly fine without the JohnnyFive/Raspi-IO. before just the Johnny five code in I had the loop just console logging the count.

dtex, 2018-11-29:
You are creating new instances of segment every time ```setDigits()``` or ```clear()``` is called. Segment should only be initialized once, and we usually do this in a ```board.on("ready")``` listener. For example:

````js
const board = new five.Board({io: new raspiio(), repl: false, debug: false});

board.on("ready", () => {
    const segment = new five.Led({pin: segPin, address: adr, controller: "PCA9685"});

   // Define your clear, clearDigits, and setDigits here
   // Make sure to remove those ```Led``` instantiations
});
````

LibbieLibbieLoo, 2018-11-29:
Hi,

I did originally have the segment instances but had 2 issues first I still had this issues and second I was unable to dynamic set the segment to be controlled in the for loop to set which segment on or off depending on bit.

dtex, 2018-11-29:
Without seeing the original code or knowing what the errors were, I'm guessing that the other problem was with variable scope.

It's important that the methods that use segment are defined within the same block where segment is intialized. This gets a little tricky when you also want to export. This is how I've handled it in the past, but I'm not sure it's the best way.

````js
const board = new five.Board({io: new raspiio(), repl: false, debug: false});
let segment; // Segment is now available throught this file

board.on("ready", () => {
     segment = new five.Led({pin: segPin, address: adr, controller: "PCA9685"});
});

module.exports = {
    setDigits: function (speed, brightness, colour, cb){
        // ...
        // segment is available now since it was created in file scope        
        segment.brightness(brightness);
        // but you have to be careful not to call setDigits() before the board ready 
        // event fires. You could create a global ```boardReady``` boolean for that.
        // ...
    }
        
};

````

ntatko, 2018-12-03:
Hi, @LibbieLibbieLoo,

I had this same issue yesterday. It turns out it was the combination of two issues:

1. I was calling my functions from an infinite loop. It would be nice to see a little of the code which you use to call `setDigits` and `clearDigits`.
2. I was trying to use `setTimeout()` to control my delays. Well, in someone's infinite wisdom, JavaScript is an asynchronous language.  It means that in this loop:
```javascript
for (let i = 0; i < 1000; i++) {
  setTimeout(() => {
    console.log("Hello from loop " + i);
  }, 250);
}
```
All of the console.logs would appear at the same time. Try it. This could very quickly overpower a Pi's computing capabilities, depending on the called function. This is why Johnny-Five ships with a `board.loop()` method, although it has no exit parameters or loop counters. 
A buddy found an npm module called [Temporal](https://www.npmjs.com/package/temporal) which can set loop delays, as well as counting the number of loops natively, giving you a fail-out condition. It stops the JavaScript's event loop, so if you have other async things happening at the same time, all of them will stop. 

Hope this is helpful

LibbieLibbieLoo, 2018-12-03:
HI @ntatko,

I ended up not using Johnny-Five in the project as I was only really using the led function for the PCA9685. I might give that a go at some point if I need to use more of the Johnny-Five functions, I now use the i2c and adafruit-pca9685 with a bit of modification.

My project now works perfectly :)

closed 2018-12-03 by LibbieLibbieLoo

godormad, 2018-12-07:
I have the same issue using raspberry pi 3 b+, even with the simplest blink example. However, older raspberry versions (pi 3b) seem to work pretty well.

dtex, 2018-12-07:
@godormad The hardware version really shouldn't matter. Can you share the code you are seeing problems with?

@ntatko Temporal doesn't stop the event loop, and it is definitely asynchronous. The thing that might be a little off-putting about it is that it checks to see if there is an action to take every single pass through the event loop. This means Temporal consumes all available processing resources and should never be used for more than 5 seconds or so in a burst.

LibbieLibbieLoo, 2018-12-07:
@godormad I could never pin down what was going on with the johnny-five stuff, I tried multiple ways round it. but think I was just trying to do some the library was not designed to do. What are you trying to do with Johnny-Five?

As @dtex said broad version shouldn't make a different. Unless you are using older version of Jessie that doesn't support pi 3 b+ fully, it worth checking this as I made this mistake a couple of months ago as I just used the image file that I download awhile ago. I have tried the johnny-five today on pi 3 b+ with a clean install of Jessie and as long as I use a `board.loop()` blinking an LED seems to be working fine.

LibbieLibbieLoo, 2018-12-07:
@godormad just had another thought, one thing that i did notice after I changed library, was that the software serial that is on the GPIO was putting a lot of weight onto the my app, are you using the GPIO serial? if you are and you can live without the Bluetooth it may be worth switching the GPIO serial to the hardware serial, this fixed issues I was having that the app would get sluggish and crash. Its worth noting though i never tried this with the Johnny-five running. if your not using the serial GPIO try disabling the the serial console in Jessie.

godormad, 2018-12-10:
@dtex As i said im using the simplest blink example. 
@LibbieLibbieLoo This is my setup: 
`Raspbian Stretch Lite, Kernel version:4.14, Release date:2018-11-13, johnny-five 1.0.0, raspi-io 9.0.0`
Disabling the mini UART seem to work. Thank you all. 
This is what ive done:
```
sudo nano /boot/config.txt
dtoverlay=pi3-disable-bt
sudo systemctl disable hciuart
```