rwaldron/johnny-five #1542: [QUESTION] Arduino Uno wifi rev2 compatibility

aldis-ameriks, 2018-12-26:
Hello,
I'm planning on getting my hands on Arduino Uno wifi rev2 for iot projects, however, I'm unsure if the board is supported.
The documentation mentions that Arduino Uno is supported, does that also include Arduino Uno Wifi rev2?

dtex, 2019-03-03:
Sorry nobody has answered here. Johnny-Five will work with any Arduino compatible device that can run firmata wifi. I do not know if that includes the Uno Wifi board. This question is probably better suited to the https://github.com/firmata/arduino repo.

Did you ever get the board?

aldis-ameriks, 2019-03-09:
@dtex no worries, and thanks for the reply. I did indeed get the board, but haven't had the time to tinker with it yet.
I've posted the question on the firmata/arduino repo - https://github.com/firmata/arduino/issues/417.

dtex, 2019-03-10:
Closing in deference to firmata/arduino#417

closed 2019-03-10 by dtex

RyderLogisticApp, 2020-08-10:
Is Arduino Wifi rev 2 still not supported?