rwaldron/johnny-five #1546: a reconnect feature ?

koraysels, 2019-01-07:
how can I look for a new board after my current board connection is lost ? I know I have to re-initialize everything and also the state but that is no problem. I just want tobe sure that if n board is found or a board disconnected, that I cankeep looking for new avaliable board and connect to it. 

I have not found any possible way to do this yet... is it even possible ?

divanvisagie, 2019-02-26:
From a 1000 foot view I'm thinking that you may be able to solve this with a package called `usb-detection`

Currently trying to get a POC working, I can get the board to init after running the app and then plugging it in, however I can't get the board to connect a second time after unplugging it.

Will keep you posted.

dtex, 2019-03-03:
@koraysels Just to be sure, you are using an Arduino connected over USB correct?

koraysels, 2019-03-04:
correct!

divanvisagie, 2019-03-22:
Right, so I have set up an example program here: https://github.com/divanvisagie/j5-restart/blob/master/index.js

With regards to:

> Currently trying to get a POC working, I can get the board to init after running the app and then plugging it in, however I can't get the board to connect a second time after unplugging it.

I found that when the board is disconnected, it wasn't removed from the `Serial.used` list so the next time you connected the board it was seen as already in the list. 

I made a change [here](https://github.com/divanvisagie/johnny-five/blob/hotplug/lib/board.js#L358) to remove the board when disconnected, however we seem to be holding onto other things as well.

When I plug the board back in there are no longer errors and the board does seem to connect again, but the LED no longer flashes.

If we make use of the `blink` callback we can see that blink is called even after the board is disconnected. This feels to me like we need to do some other cleanup as well but I have no idea where to start in terms of this.

LiorBird, 2020-01-31:
@divanvisagie @koraysels thanks for all of your input but I was curious if you ever managed to fix that problem? I'm trying to use an Arduino Due using the Johnny-Five script function and I encounter the same problem... Sorry for bringing up this issue again, but I couldn't find an answer anywhere else. Thank you so much!

closed 2022-01-09 by stale[bot]