rwaldron/johnny-five #1548: Arduino Uno thermometer DS18B20 and ConfigurableFirmata error with NodeJs 10

devsheder, 2019-01-17:
Hi,
I've tried to read temperature from DS18B20 sensor using this doc : [http://johnny-five.io/examples/temperature-ds18b20/](http://johnny-five.io/examples/temperature-ds18b20/)

First of all, I've uploaded this firmware on my Arduino Uno : [https://github.com/firmata/ConfigurableFirmata/blob/master/examples/ConfigurableFirmata/ConfigurableFirmata.ino](https://github.com/firmata/ConfigurableFirmata/blob/master/examples/ConfigurableFirmata/ConfigurableFirmata.ino)

And then tried to run the snippet in the doc and I got this error :
```
1547733530197 Available /dev/tty.usbmodem1411
1547733530215 Connected /dev/tty.usbmodem1411
1547733533998 Repl Initialized
>> Board is ready
events.js:167
      throw er; // Unhandled 'error' event
      ^
Error: 1-Wire device search timeout - are you running ConfigurableFirmata?
    at Timeout._onTimeout (/Users/user/projects/TemperatureSensorIoT/node_modules/firmata/lib/firmata.js:1485:14)
    at listOnTimeout (timers.js:324:15)
    at processTimers (timers.js:268:5)
Emitted 'error' event at:
    at EventEmitter.<anonymous> (/Users/user/projects/TemperatureSensorIoT/node_modules/johnny-five/lib/thermometer.js:185:18)
    at Timeout._onTimeout (/Users/user/projects/TemperatureSensorIoT/node_modules/firmata/lib/firmata.js:1485:5)
    at listOnTimeout (timers.js:324:15)
    at processTimers (timers.js:268:5)
```

I'm sure the sensor is working because I've tested it with this code [https://create.arduino.cc/projecthub/TheGadgetBoy/ds18b20-digital-temperature-sensor-and-arduino-9cc806](https://create.arduino.cc/projecthub/TheGadgetBoy/ds18b20-digital-temperature-sensor-and-arduino-9cc806) and got this :
[image]

Note that the NodeJS code is running on a Mac laptop.

Thanks a lot for any help

Sheder

dtex, 2019-01-17:
Hi @devsheder ,

That configurableFirmata sketch has to be modified to include oneWire. An easier way to get the exact configurableFirmata you need is to build it over at http://firmatabuilder.com

devsheder, 2019-02-04:
Hi @dtex ,
Thanks for your reply.
I don't know how but I fixed my problem.
I've generate a Firmata file with all the core features. Then I've installed it on my Arduino : it wasn't working with J5.
Then I reinstalled this code : https://github.com/firmata/ConfigurableFirmata/blob/master/examples/ConfigurableFirmata/ConfigurableFirmata.ino and It worked !
If you have an explanation I'm curious but if you have not it's ok.

Thx !

closed 2019-02-04 by devsheder