rwaldron/johnny-five #1549: esc.throttle is not a function

bart, 2019-01-22:
Hey everyone,

I'm trying to control a brushless motor via an ESC using J5 version 1.0. But when executing my super simple code I'm getting `TypeError: esc.throttle is not a function`

My code looks like this:

```
var five = require("johnny-five"),
    board = new five.Board();

board.on("ready", function() {
  var esc = new five.ESC(9);

  esc.throttle(1500);
});
```

Any idea what's wrong here?

Thanks!

mkirank, 2019-03-02:
I ran into this issue and it looks like throttle does not exist or has been removed. The way to set this is to use esc.speed()

dtex, 2019-03-02:
@rwaldron I'll update the wiki but have a question. 

For directional motors It appears that the speed method scales the speed param from PWM min to PWM max w/o regard to direction or neutral point so:

````js
// esc.neutral == 50
esc.speed(0); // full reverse
esc.speed(25); // reverse at 50%
esc.speed(50); // stopped 
esc.speed(75); // forward at 50%
esc.speed(100); // forward at 100%
````

I don't have an ESC to test it out.

HipsterBrown, 2019-08-13:
The wiki has been updated with the new APIs merged in #1484. So this issue can probably be closed now.

closed 2019-08-13 by rwaldron