rwaldron/johnny-five #1551: expander mcp23017 wrong schema

fair-man, 2019-01-27:
[image]

fair-man, 2019-01-27:
http://www.alldatasheet.com/view.jsp?Searchword=Mcp23017&gclid=CjwKCAiAyrXiBRAjEiwATI95mYXjwjKSzWBdQqTwzhMq-mvXwNO-PbqrzpH0phK4swdmLGKJ6P_3cxoCnesQAvD_BwE

dtex, 2019-01-27:
Thanks for the catch. Would you like to submit a PR? It should just be a matter of updating the fzz and png files.

PeterArriaza, 2019-10-03:
is this still open? If so, I'd be happy to fix this!

dtex, 2019-10-04:
That would be great.

shawnpetros, 2019-10-12:
Haven't seen any movement on this in a week, if no one's claimed I can take a shot at it...I'll reference in a PR if I can get it done in the next day or so...

dtex, 2019-10-12:
Thanks for the reminder. I need to close this.

closed 2019-10-12 by dtex