rwaldron/johnny-five #1554: led rgb and smd rgb not working

darkworks, 2019-02-13:
hi i have led rgb and smd rgb i tried to run default example code on it through my arudio mega 2560 board and i found that its not working , 

```
    var led = new five.Led.RGB({
      pins: {
        red: 34,
        green: 38,
        blue: 36
      }
    });

  // Add led to REPL (optional)
  this.repl.inject({
    led: led
  });

  // Turn it on and set the initial color
  led.on();
  led.color("#FF0000");

  led.blink(1000);
```

and this one is working but only white light cant set rgb value 

```
    var rgb_red_led  = new five.Pin(34);   // Red LED
    var rgb_blue_led  = new five.Pin(36);  // Blue LED
    var rgb_green_led  = new five.Pin(38);  // Green LED
    rgb_red_led.write(100);
    rgb_green_led.write(200);
    rgb_blue_led.write(25);
```

dtex, 2019-02-13:
Hi @darkworks can you send us a picture of your wiring?

darkworks, 2019-02-13:
ok sure
main LED picture

[image]

LED on breadBoard , gray is ground ,  blue is blue , green is green and purple is red

[image]

digital pins : red = 34 , blue 36 , green 38  

[image]

dtex, 2019-02-13:
So the wiring looks right. What's that shield? Is that where the ground wire attaches (I don't see it in your Arduino pic)? Is it possible there is a separate ground circuit on that shield and that's what your grey wire is plugged into?

Based on the behavior, it sounds like 34, 36, and 38 are not handling PWM signals, though I'm pretty sure they should. Is it possible you have a REALLY old version of firmata on there? One quick test would be to hook the RGB pins up to 8, 10, 12 and see if it works there. At least then we'd know if we are on the right track.

darkworks, 2019-02-13:
hmmmm ya right its CNC shield am taking ground and v5 from it to bread board  , i have flashed the board with latest version of https://github.com/firmata/ConfigurableFirmata  ,

darkworks, 2019-02-13:
i directly insert the rgb led in mega2560 GND 13,12,11 pins   , now its on but only green ,

dtex, 2019-02-13:
Which includes did you uncomment from the .ino file? Or did you use firmataBuilder?

darkworks, 2019-02-13:
i download ConfigurableFirmata library into Arduino IDE from sketch menu > include library menu , after that went to FILE menu > examples > ConfigurableFirmata > ConfigurableFirmata  and then uploaded it to board

dtex, 2019-02-13:
Ah, that's the problem. You have to [enable the features you need in configurableFirmata](https://github.com/firmata/ConfigurableFirmata#usage). 

The easier way is to create a custom .ino at http://firmatabuilder.com

Are you sure you need configurable? I know nothing about that CNC shield but if you are only using configurable for stepper support, I think this shield controls the steppers for you and you'd be fine with standardFirmataPlus.

darkworks, 2019-02-13:
ok i will try the builder , ya i have stepper motors , initially i had standardFirmdata plus on board but stepper motor was not working , also http://johnny-five.io/api/stepper/   here its mentioned to flush board with advance firm data , i used advance firm data and then was able to operate motors and then later i came across ConfigurableFirmata

darkworks, 2019-02-13:
no success just flush board with standardFirmData plus and  when run 

```
    var led = new five.Led.RGB({
      pins: {
        red: 13,
        green: 12,
        blue: 11
      }
    });

  // Add led to REPL (optional)
  this.repl.inject({
    led: led
  });

  // Turn it on and set the initial color
  led.on();
  led.color("#eddc47");
```

its still white color :(

darkworks, 2019-02-24:
fixed it not sure how but now working on had to use  PWM pins

closed 2019-02-24 by darkworks