rwaldron/johnny-five #1559: DMM drives with johnny five

darkworks, 2019-02-26:
i have Question can we use johnny five with DMM DYN4 AC serve drives 

http://www.dmm-tech.com/Files/DYN4MS-ZM7-A10A.pdf

thanks

dtex, 2019-03-03:
Not out of the box. What board are you using? Some of the IO plugins support serial so the DYN232M should be able to be implemented.

I've added this to the [Requested Features](https://github.com/rwaldron/johnny-five/wiki/Requested-Features) page.

closed 2019-03-03 by dtex

darkworks, 2019-03-04:
am using Arduino Mega 2560  , also i have i2c module PCF8574T

darkworks, 2019-03-05:
would it will be possible to user johnny five and Serial port : https://serialport.io/docs/en/guide-usage     together

dtex, 2019-03-06:
Johnny-Five already uses Serial port as a transport for firmata when talking to an Arduino Mega. You can use a second UART on the Mega to talk to the servos using functionality that is already available.

The serial methods that are available are documented at https://github.com/firmata/firmata.js 

The most practical way to do this would be to add a new device and/or controller to the Johnny-Five servo class. If these servos have some additional feature beyond a regular servo (i.e. position encoding) then you would need a new device. If they are programmatically indistinguishable from standard hobby servos then you're fine with the existing default device.

Either way, you will need to add a new controller that uses serial to send messages. The only Johnny-Five class that currently uses serial is GPS so you could crib some stuff from there.

darkworks, 2019-03-11:
thank you very much :) , i will play with it once i get dmm drives , hopefully i next month , thanks for the direction , i will update you how it goes ,