rwaldron/johnny-five #1561: Change pin of proximity dynamically

sansil, 2019-03-26:
Hi!

I'm working in a scratch extension and I want to change the pin dynamically of a proximity sensor.
How can i do that?

Thank you in advance!

dtex, 2019-03-26:
Johnny-Five wasn't designed with that in mind.

I'm always reluctant to tell someone something isn't possible, because given enough time and determination most things are, but there will be some challenges. The specific challenges vary by device. What kind of proximity sensor are you using?

sansil, 2019-03-26:
Hi dtex, thank you for your response.

I'm using HC-SR04 proximity sensor. I tried to remove listeners of the proximity sensor instance and then create a new one but not work at all.

dtex, 2019-05-18:
Hi @snasil Sorry for the slow response. Have you tried doing this from node w/o the Scratch layer? Can you share that code and the error you are getting (if any)?

closed 2019-08-15 by dtex