rwaldron/johnny-five #1562: mixed model AVR2.DLL failed to authorize

Hamza13102, 2019-03-27:
Hi, kindly assist am getting this error when running a simulation in proteous

PROSPICE 8.04.00 (Build 21720) (C) Labcenter Electronics 1993-2016.
mixed model AVR2.DLL failed to authorize - Missing or invalid Customer Key.. [ARD1]
mixed model PIC16EX.DLL failed to authorize - Missing or invalid Customer Key.. [SONAR1]
mixed model PIC16EX.DLL failed to authorize - Missing or invalid Customer Key.. [SONAR2]
Net VSS taken as alias for GND
Loaded netlist 'C:\Users\HAMZA\AppData\Local\Temp\LISA8415.SDF' for design 'arduino two_way.pdsprj'
Real Time Simulation failed to start.
Simulation FAILED due to fatal simulator errors.

dtex, 2019-03-27:
Hi @Hamza13102 

Take a look at this closed issue comment:

https://github.com/rwaldron/johnny-five/issues/314#issuecomment-38364232

closed 2019-05-18 by dtex

prasenjit9862, 2020-07-19:
> Hi, kindly assist am getting this error when running a simulation in proteous
>
> PROSPICE 8.04.00 (Build 21720) (C) Labcenter Electronics 1993-2016.
> mixed model AVR2.DLL failed to authorize - Missing [...]

Sir I have also same problem please help me

prasenjit9862, 2020-07-19:
How can I'm solving this problem please help me

dtex, 2020-07-19:
@prasenjit9862 Did you follow the advice in https://github.com/rwaldron/johnny-five/issues/314#issuecomment-38364232 ?