rwaldron/johnny-five #1567: Any Infinite loop wouldn't execute

DoomScripty, 2019-04-09:
Using J5 , to setup a servo motor I wanted to have a faster alternative to
 `Servo.sweep(interval)` 
as I wanted my Servo motor to go from 70° to 90° back to 70° each Interval T . 

Doing an infinite loop

 `while(true){ 
servo.to(70); 
setTimeout(function() {
servo.to(90);}, 300); 
}
`

It would either execute once and stop or didn't execute at all. 
My Improvised solution was a recursive setTimout Function whit Interval T= 1 second

`function repeatingFunc() {
    console.log("It's been 1 second. Execute the function again.");
	
    

  servo.sweep({ range: [70, 90], interval: 300 });
  setTimeout(function() {
servo.stop();}, 300);

 
	setTimeout(repeatingFunc, 1000);}

setTimeout(repeatingFunc, 1000);`

dtex, 2019-04-09:
This should be all you need:

````js
servo.sweep({ range: [70, 90], interval: 300 });
````

Sweep just keeps on running until you call ```servo.stop();```

DoomScripty, 2019-04-09:
Thanks for the fast reply, 

This surely does fix the servo.sweep being slow, the issue is we want to repeat this action each 10 seconds for example and want this to run for ever 

example " servo sweeps, holds for 10 sec, servo sweeps, holds for 10 sec,....." for ever 

and we couldn't do it by infinite loops for a unknown reason.

dtex, 2019-04-10:
Your loops were stepping on each other. In your first example, you were creating infinite timeouts within your while loop.

You can accomplish what you've described with a custom animation:

````js
var five = require("johnny-five");
var board = new five.Board();

board.on("ready", function() {
  var servo = new five.Servo(9);
  var animation = new five.Animation(servo);

  // Create an animation segment object
  let mySweep = function() {
    animation.enqueue({
      duration: 300,
      cuePoints: [0, 0.5, 1.0],
      keyFrames: [ 70, 90, 70]
    });
    setTimeout(mySweep, 10300);
  }

  mySweep();

});
````

closed 2019-05-18 by dtex