rwaldron/johnny-five #1569: How to get available boards ?

Yvanovitch, 2019-04-15:
Hi,
I want to create a list of available boards without knowing how many there are before.
It can go from 0 to 8+.
I found this ticket [1456](https://github.com/rwaldron/johnny-five/pull/1456) refering to Board.detect but it seems it's been moved to Serial.detect, but it seems Serial isn't exposed, is it ?

When I use Boards, I still need to provide an array of "{repl: false }" (because I don't want repl).

When I create the first Board, I get a message with class: "Available" containing "COM8,COM7". I can parse this message to know how many boards there are, but is there a cleaner way ? 

In addition I have to create at least one Board in order to get this information, whereas I'd prefer to create it only if there is one.
Thanks

dtex, 2019-04-15:
Are these boards running firmata?

Yvanovitch, 2019-04-15:
Yes, and I can read/write each of them

dtex, 2019-04-15:
Sorry, I should have asked if they are connecting through serialport (USB).

dtex, 2019-04-15:
If they are, that is provided by serialport. You can require that package directly and use ```list()``` (It returns a promise)

````js
const SerialPort = require('serialport');
SerialPort.list()
  .then(ports => {
    const port = ports.find(port => /arduino/i.test(port.manufacturer));
    if (!port) {
      console.error('Arduino Not found');
      process.exit(1);
    }
    console.log(port.comName);
  });
````

Yvanovitch, 2019-04-15:
Ha, that solves my problem. I was looking for something inside J5 but I guess it wasn't relevant to duplicate this code.
Thanks!

closed 2019-04-15 by Yvanovitch