rwaldron/johnny-five #1570: Undocumented breaking change in Servo module

hegelstad, 2019-05-01:
The commit linked to below introduces an undocumented breaking change. At least some IO libraries expect the servoWrite function to receive degrees and not microseconds as it now does in the current release.

https://github.com/rwaldron/johnny-five/commit/c33155e5e8eb6d8a1f594793ec64655aa6175d06#diff-2ea177c952d168875093e55f1f6c6991R81

For instance, the excellent library [pi-io](https://github.com/fivdi/pi-io) relies on degrees to write the right value to the Pin.

dtex, 2019-05-01:
It is [documented here](https://github.com/rwaldron/io-plugins#servowritepin-value).

It looks like linux-io (the foundation for pi-io) was [patched here](https://github.com/fivdi/linux-io/commit/d7b4ee458f494ad4f31be15dee73cdd4f505cb24) and the right thing to do would be to fix the behavior between pi-io and linux-io.

I made an effort to reach out to all the io-plugin maintainers and patch their code before this ever landed. pi-io wasn't caught up in that sweep because well, I'd never heard of it (sorry @fivdi).

fivdi, 2019-05-01:
This has been fixed with pi-io@1.0.1 so this issue can be closed.

closed 2019-05-01 by dtex

hegelstad, 2019-05-02:
Thanks for clarifying, I didn't know about the io-plugins repository 👀