rwaldron/johnny-five #1572: Extend Raspberry PI with Arduino Pro Mini

pirquessa, 2019-05-09:
Hi,

I'm making a robot with a Raspberry Pi Zero W. Works so well ! Thanks for this amazing lib.

I now need some more PWM, Analog input and things like that. I'm planning to use an [Arduino Pro Mini](https://store.arduino.cc/arduino-pro-mini) to get plenty more IO. My first plan is to connect both boards with I2C, have a dedicated C program in the Arduino to interpret I2C communication and orders.

But I realize that it will be cleaner / easier to control the Arduino with the johnny-five main program ! If I understand well, the only way to do that is to connect both board with USB... But there is no USB in the Arduino Pro Mini... **Is there any other solution ?** (I would prefer a more compact solution of communication, like the only two wire of I2C)

Tanks again :)

dtex, 2019-05-09:
Use one or more PCA-9685 expanders. You can chain many of these on I2C and Johnny-Five already supports them:
 [At Digikey](https://www.digikey.com/catalog/en/partgroup/adafruit-16-channel-12-bit-pwm-servo-driver-i-c-interface/49477?utm_adgroup=xGeneral&slid=&gclid=Cj0KCQjwn8_mBRCLARIsAKxi0GJJ-p7mWTNLm8OKneJ4C_MSCbX6qOiAv5KJ6U90-q7g8ydzI4yrsXkaAiJ4EALw_wcB)

pirquessa, 2019-05-09:
That's nice ! But it only manage servo (PWM), no Analogic input.... AND I already have one Arduino Pro Mini :-p

I see the documentation for current expanders [here](http://johnny-five.io/api/expander/). Maybe I could do some kind of expander implementation for my Arduino ? Where is the code of the expander you mention (PCA9685) ?

dtex, 2019-05-09:
Could you just use something like this to give you a USB plug on your Mini?

https://store.arduino.cc/usa/usb-serial-converter

pirquessa, 2019-05-09:
Sure I can, but it turns a micro Arduino into a big Arduino... And take the only USB port of the RPI Zero... :-/

dtex, 2019-05-18:
Well, there's plenty of ways to accomplish what you want to do, but they all involve either switching out some hardware or doing a lot of work on the software. I'm shooting from the hip, but I think either of these suggestions could work...

1) Swap out the Arduino Pro Mini with an ESP32 you could just connect it as a second board in Johnny-Five wirelessly.

2) If you create an I2C transport for firmata that would allow you to chain multiple boards and do everything you can do with firmata. Each one would just be a separate board like the above suggestion. Note that this would be a LOT of work.

pirquessa, 2019-05-19:
First solution seems not very optimal for battery. Second is what I need! I'll try something cheaper... :)

dtex, 2019-05-20:
Regarding option 1, I need to correct myself. Firmata does not yet support that device. An ESP8266 with wifi would work.

I see you're going with option 2. I can't help but think that is the more difficult option but I wish you luck

closed 2019-05-20 by dtex