rwaldron/johnny-five #1573: Error trying to control multiple thermometers

communque, 2019-05-15:
**Devices involved:**
- 2 [url=https://www.adafruit.com/product/2471]Huzzah Breakout ESP8266[/url]s
- 2 [url=https://www.adafruit.com/product/1782]MCP9808 Thermometer[/url]s
- 1 CPU controlling both using Johnny-Five.

The following code works fine when there's one thermometer, but it fails trying to control two.
**In other words this code works when line 4 is commented out and fails when it's included.**

        var five = require("johnny-five");
        var {EtherPortClient}=require("etherport-client");
        var Thermometers=[
        	//{Name:"Thermometer1", Ip:"192.168.1.101"}, //Line 4 -- Uncommenting this line will cause the code to fail.
        	{Name:"Thermometer2", Ip:"192.168.1.102"} 
        ];
        TrackThermometers();
        
        function TrackThermometers(){
        	Thermometers.forEach(function(ThisThermometer, ThermometerCount){
        		ThisThermometer.Board=new five.Board({
        			port: new EtherPortClient({
        				host: ThisThermometer.Ip,
        				port: 3030
        			}),
        			repl: false
        		});
        		ThisThermometer.Board.on("ready", function(){
        			ThisThermometer.Controller=new five.Thermometer({ //!!This cmd triggers the error
        				controller:"MCP9808"
        			});
        			ThisThermometer.Controller.on("change", function(){
        				console.log(this.id, this.fahrenheit);
        			});
        		})
        	});
        }

The only difference is whether or not two or more thermometers are included.
Error is: _"node_modules/johnny-five/lib/board.pins.js:91  var type = board.pins.type;  TypeError: Cannot read property 'type' of null"_'
It seems to be the result of trying run _new five.Thermometer({controller:"MCP9808"})_ more than once.

dtex, 2019-05-17:
I'm just guessing here, but the problem might be that your second call to ```new five.Thermometer()``` does not specify which board to use. You can try the code below to see if that helps (I only added the line specifying the board). If that doesn't change anything, can you post the full stack trace?

````js
var five = require("johnny-five");
var {EtherPortClient}=require("etherport-client");
var Thermometers=[
  //{Name:"Thermometer1", Ip:"192.168.1.101"}, //Line 4 -- Uncommenting this line will cause the code to fail.
  {Name:"Thermometer2", Ip:"192.168.1.102"} 
];
TrackThermometers();

function TrackThermometers(){
  Thermometers.forEach(function(ThisThermometer, ThermometerCount){
    ThisThermometer.Board=new five.Board({
      port: new EtherPortClient({
        host: ThisThermometer.Ip,
        port: 3030
      }),
      repl: false
    });
    ThisThermometer.Board.on("ready", function(){
      ThisThermometer.Controller=new five.Thermometer({
        board: thisThermometer.Board,
        controller:"MCP9808"
      });
      ThisThermometer.Controller.on("change", function(){
        console.log(this.id, this.fahrenheit);
      });
    })
  });
}
````

communque, 2019-05-17:
Thank you Donovan!  That nailed it.
specifying the parameter 'board' in new five.Thermometer is exactly what I was looking for.

I hadn't been able to find much documentation about 'Thermometer' props.  Only 'controller' and 'pin' so far, and now 'board', thanks to you.  Are there more props to consider / is there documentation on that?

dtex, 2019-05-17:
Have you been to http://johnny-five.io/api/thermometer/

The ```board``` option is only documented in the Boards API

communque, 2019-05-17:
> Have you been to http://johnny-five.io/api/thermometer/

That's where you'll find ample examples of the 'controller' property, as well as 'pin', which is apparently necessary / applicable to certain thermometers.  I'm using MCP9808, which doesn't seem to have any use for 'pin'.

How did you guess at 'board' being a property of distinction in 'Thermometer'

dtex, 2019-05-17:
I was guessing about the cause, not the existence of the board param. I had seen it in one of the multi-board examples.

dtex, 2019-05-18:
I've added #1575 tp get the docs updated so I'm going to close this issue.

closed 2019-05-18 by dtex

communque, 2019-05-18:
Thanks again!