rwaldron/johnny-five #1575: Update docs to include `board` param

dtex, 2019-05-18:
There is an optional ```board``` param than may be passed in the constructor opts for all device classes. Simply put, it allows the user to specify which board they want to use in projects that have multiple boards. It is currently not documented anywhere and should probably be added to all the classes.

leonlafa, 2019-06-18:
More than happy to update the docs if this is still a requirement.

dtex, 2019-06-18:
That would be great. Make sure you note how the docs are built. It's covered in the last paragraph of the [writing documentation section](https://github.com/rwaldron/johnny-five/blob/master/CONTRIBUTING.md#writing-documentation)

leonlafa, 2019-06-19:
@dtex I will make sure to follow the documents.  
I do have a couple of questions before I begin:

1) What is the property name, type, value, and description of the param?
2) Is this param exclusive to the board component class?

dtex, 2019-06-19:
The param name is "board" and it's available to all the device classes (LED, motor, GPS, whatever). Note that Animation, Board, Boards, and Fn are not device classes. 

You can only pass a board instance. 

It's only necessary when you are using multiple boards. The default value is the first board mounted.

The description would be something like "The board instance your LED* is connected to".  *Or whatever the device is.

I propose editing one class wiki page and running it by @rwaldron before updating them all. In retrospect, you may not need to worry about my warning to note how the docs are built. This will all be in the wiki.

leonlafa, 2019-06-19:
@rwaldron I have updated the LED wiki page.  Please review before I amend any more pages.

leonlafa, 2019-06-21:
> @rwaldron I have updated the LED wiki page. Please review before I amend any more pages.

Anyone had a chance to review yet? @dtex @rwaldron

dtex, 2019-06-21:
I have, and it looks fine to me.

leonlafa, 2019-06-21:
@dtex okay that's good I'll commit to going through all of the wikis this weekend and doing the same.

ibrahim-sowunmi, 2019-10-23:
Has this been done?

Bazel7, 2019-11-07:
Super

closed 2020-11-14 by stale[bot]