rwaldron/johnny-five #1576: Metro Mini ttyusb0 issue

thelaplage, 2019-05-19:
Hi, I'm having an issue with some working Johnny-five code that works with an arduino uno (on port ttyACM0), I can't get it to work with a Metro Mini (port ttyUSB0), but i'm not getting any errors so i'm having a difficult time diagnosing the problem. It runs, connects, but doesn't respond to any javascript commands. I feel like that means it's a socket.io issue, but I've found nothing on this issue.

I'm using the johnny-five stepper motor code and configurable firmata. And again, the code works as intended with an Uno connected, javascript controlling a stepper.

With the Metro mini, my node script runs and connects with no errors! 
Socket.io recognizes the Metro and, says port is available, and connects to ttyUSB0. I get all the right debug info, down to "websocket writing 1"
But then I get nothing back when I actually try to interact with it!
I'm expecting something with console.log when I emit, but nothing. But exact same code works on the Uno!

I feel like I must be missing something very obvious. I've reuploaded configurable firmata. The Metro Mini is supposed to be treated like an Uno in the Arduino IDE. Any guesses as to what this could be? Thanks!

dtex, 2019-05-19:
It sounds like there is a lot more going on than just Johnny-Five talking to the Metro (socket.io, console.log, specifying different USB ports). Have you tried just blinking the LED on pin 13 straight from node, without all the websocket stuff?

thelaplage, 2019-05-19:
Hi, thanks for reply
It was an issue with ConfigurableFirmata on the Metro
I can confirm that stepper-driver.js works on the Metro with AdvancedFirmata

here's details for anybody in similar boat that wants to reference:

---

Here's what I get from running stepper-driver.js example  (http://johnny-five.io/examples/stepper-driver/) with ConfigurableFirmata on the Metro

```
pi@raspberrypi:/var/www/html/0515 $ node stepper-driver.js
1558279453008 Available /dev/ttyUSB0  
1558279453056 Connected /dev/ttyUSB0  
1558279458114 Repl Initialized  
```
and then nothing in the console... 

Running that exact same code on the exact same Pi, but with AdvancedFirmata gives me:

```
pi@raspberrypi:/var/www/html/0515 $ node stepper-driver.js
1558280628303 Available /dev/ttyUSB0  
1558280628347 Connected /dev/ttyUSB0  
1558280632141 Repl Initialized  
>> Done moving CCW
Done moving CW
```

Not sure why, but there seems to be an issue with using a Metro Mini and ConfigurableFirmata

dtex, 2019-05-19:
Oh, interesting. Are you using http://firmatabuilder.com/ ?

closed 2020-08-16 by stale[bot]