rwaldron/johnny-five #1577: Missing serial port dependency in J5 v1.1.0

djipco, 2019-05-24:
I have been using johnny-five v1.0.0 within NW.js projects for quite a while now. However, since version 1.1.0, my projects are giving me this error:

```Uncaught Missing serialport dependency```

If I switch back to v1.0.0, everything works fine again. 

Here's how you can reproduce this problem on Windows. Simply [download NW.js SDK](https://nwjs.io/downloads/) and decompress it. Inside the decompressed folder, create a subfolder called `package.nw`. In this folder, put the 2 files below and double-click the ```nw.exe``` icon (right-click to open dev tools and view the error)

**package.json**

```json
{
  "name": "test",
  "main": "index.html",
  "dependencies": {
    "johnny-five": "1.1.0",
    "nwjs-j5-fix": "^1.0.3"
  }
}
```

**index.html**
```html
<!DOCTYPE html>

<html lang="en">

  <body>
    <script>
        const five = require("johnny-five");
        require("nwjs-j5-fix").fix();
        let uno = new five.Board({port: "COM4"});
        uno.on("ready", () => console.log("Ready!"));
    </script>
  </body>

</html>

```

I tried this on macOS and Windows 10 and it throws the same error in both cases. I tried manually installing the **serialport** module and it does not change anything. Ideas?

dtex, 2019-05-24:
The bump to 1.1.0 requires a newer version of serialport and that seems to be the problem here. I know sometimes serialport can be a booger with nw, but I don't have personal experience with it. Have you tried asking about serialport v7 in nw over in the [serialport repo](https://github.com/serialport/node-serialport)? That's where you would get an informed answer.

My hacky solution (maybe) would be to remove all occurrences of serialport in node_modules and just install v6.  I don't think the SerialPort API has changed in a way that would break firmata.js/johnny-five

djipco, 2019-05-24:
Thanks for your response @dtex but I'm not sure I follow. Which serialport version is required by J5 v1.1.0 ? Does it need to be installed separately?

dtex, 2019-05-24:
Johnny-Five v1.1.0 has seriaport^7.1.5 under optional dependencies.
Prior to that, it was serialport^6.0.4

djipco, 2019-05-24:
Even when I manually install serialport 7.1.5, the problem remains. I've filed [an issue](https://github.com/serialport/node-serialport/issues/1878) in the serialport repo as well. I don't even know where the problem lies. Is it with J5, serialport or NW.js?!

P.S. When Johnny-Five runs within NW.js, it shouldn't even use the serialport module since it has access to the built-in `chrome.serial` API. I'm not sure how to tell Johnny-Five to use that, though.

reconbot, 2019-05-24:
This is a weird error. There was a change in our binary modules between those two versions. We moved all binary stuff into the `@serialport/bindings` package. So if there's any rebuilding going on that targets a specific package, that might need to be updated.

djipco, 2019-05-24:
I'm not sure what you are asking...

reconbot, 2019-05-24:
I don't know enough about nwjs. 
Can you confirm that if you use the nwjs repl, that you can require `serialport` and `@serialport/bindings`?

djipco, 2019-05-27:
As far as I know, there is no REPL for NW.js. When I try to require `serialport` I get an error regarding a Node.js version mismatch. 

What I do not understand, is that none of this was an issue with v1.0.0 of Johnny-Five...

Here's the full error:

```
Uncaught Error: The module '/private/var/folders/9t/vtkj76ls0vq3vbps1qydzl7m0000gn/T/AppTranslocation/B49419CB-8847-45F4-B0CC-AB52E24C9E71/d/nwjs.app/Contents/Resources/app.nw/node_modules/@serialport/bindings/build/Release/bindings.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 64. This version of Node.js requires
NODE_MODULE_VERSION 72. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:861)
    at Module.load (internal/modules/cjs/loader.js:666)
    at Function.Module._load (internal/modules/cjs/loader.js:581)
    at Module.require (internal/modules/cjs/loader.js:706)
    at require (internal/modules/cjs/helpers.js:16)
    at bindings (private/var/folders/9t/vtkj76ls0vq3vbps1qydzl7m0000gn/T/AppTranslocation/B49419CB-8847-45F4-B0CC-AB52E24C9E71/d/nwjs.app/Contents/Resources/app.nw/node_modules/bindings/bindings.js:112)
    at Object.<anonymous> (private/var/folders/9t/vtkj76ls0vq3vbps1qydzl7m0000gn/T/AppTranslocation/B49419CB-8847-45F4-B0CC-AB52E24C9E71/d/nwjs.app/Contents/Resources/app.nw/node_modules/@serialport/bindings/lib/darwin.js:1)
    at Module._compile (internal/modules/cjs/loader.js:799)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:816)
    at Module.load (internal/modules/cjs/loader.js:666)
```

djipco, 2019-07-10:
Okay, the version mismatch error was because the package needed to be recompiled against the correct version of Node. This has been done. 

However, when I try to use NW.js v0.39.2 with Johnny-Five 1.1.0 (requiring serialport 7.1.5), NW.js simply [crashes](https://github.com/nwjs/nw.js/issues/7107). If I revert to Johnny-Five 1.0.0 (everything else being equal), it works just fine.

This is what `npm ls` looks like in my test project:

```
├─┬ johnny-five@1.1.0
│ ├── browser-serialport@2.1.0
│ ├─┬ chalk@2.1.0
│ │ ├─┬ ansi-styles@3.2.1
│ │ │ └─┬ color-convert@1.9.3
│ │ │   └── color-name@1.1.3
│ │ ├── escape-string-regexp@1.0.5
│ │ └─┬ supports-color@4.5.0
│ │   └── has-flag@2.0.0
│ ├── color-convert@1.2.2
│ ├── ease-component@1.0.0
│ ├─┬ firmata@2.0.0
│ │ ├── firmata-io@2.0.0
│ │ └── serialport@7.1.5 deduped
│ ├── lodash.clonedeep@4.5.0
│ ├── lodash.debounce@4.0.8
│ ├── nanotimer@0.3.10
│ ├─┬ serialport@7.1.5
│ │ ├─┬ @serialport/binding-mock@2.0.5
│ │ │ ├─┬ @serialport/binding-abstract@2.0.5
│ │ │ │ └── debug@4.1.1 deduped
│ │ │ └── debug@4.1.1 deduped
│ │ ├─┬ @serialport/bindings@2.0.8
│ │ │ ├── @serialport/binding-abstract@2.0.5 deduped
│ │ │ ├── @serialport/parser-readline@2.0.2 deduped
│ │ │ ├─┬ bindings@1.5.0
│ │ │ │ └── file-uri-to-path@1.0.0
│ │ │ ├── debug@4.1.1 deduped
│ │ │ ├── nan@2.14.0
│ │ │ └─┬ prebuild-install@5.3.0
│ │ │   ├── detect-libc@1.0.3
│ │ │   ├── expand-template@2.0.3
│ │ │   ├── github-from-package@0.0.0
│ │ │   ├── minimist@1.2.0
│ │ │   ├─┬ mkdirp@0.5.1
│ │ │   │ └── minimist@0.0.8
│ │ │   ├── napi-build-utils@1.0.1
│ │ │   ├─┬ node-abi@2.9.0
│ │ │   │ └── semver@5.7.0
│ │ │   ├── noop-logger@0.1.1
│ │ │   ├─┬ npmlog@4.1.2
│ │ │   │ ├─┬ are-we-there-yet@1.1.5
│ │ │   │ │ ├── delegates@1.0.0
│ │ │   │ │ └─┬ readable-stream@2.3.6
│ │ │   │ │   ├── core-util-is@1.0.2
│ │ │   │ │   ├── inherits@2.0.4
│ │ │   │ │   ├── isarray@1.0.0
│ │ │   │ │   ├── process-nextick-args@2.0.1
│ │ │   │ │   ├── safe-buffer@5.1.2 deduped
│ │ │   │ │   ├─┬ string_decoder@1.1.1
│ │ │   │ │   │ └── safe-buffer@5.1.2 deduped
│ │ │   │ │   └── util-deprecate@1.0.2
│ │ │   │ ├── console-control-strings@1.1.0
│ │ │   │ ├─┬ gauge@2.7.4
│ │ │   │ │ ├── aproba@1.2.0
│ │ │   │ │ ├── console-control-strings@1.1.0 deduped
│ │ │   │ │ ├── has-unicode@2.0.1
│ │ │   │ │ ├── object-assign@4.1.1
│ │ │   │ │ ├── signal-exit@3.0.2
│ │ │   │ │ ├─┬ string-width@1.0.2
│ │ │   │ │ │ ├── code-point-at@1.1.0
│ │ │   │ │ │ ├─┬ is-fullwidth-code-point@1.0.0
│ │ │   │ │ │ │ └── number-is-nan@1.0.1
[... 67 lines cut ...]
 7: 0x11913fcf6 v8::internal::JSPromise::Fulfill(v8::internal::Handle<v8::internal::JSPromise>, v8::internal::Handle<v8::internal::Object>) [/Users/jpcote/Desktop/TESTNWJS/nwjs0.39.2.app/Contents/Versions/75.0.3770.100/nwjs Framework.framework/Versions/A/nwjs Framework]
 8: 0x1192bebb4 v8::internal::Runtime::SetObjectProperty(v8::internal::Isolate*, v8::internal::Handle<v8::internal::Object>, v8::internal::Handle<v8::internal::Object>, v8::internal::Handle<v8::internal::Object>, v8::internal::StoreOrigin, v8::Maybe<v8::internal::ShouldThrow>) [/Users/jpcote/Desktop/TESTNWJS/nwjs0.39.2.app/Contents/Versions/75.0.3770.100/nwjs Framework.framework/Versions/A/nwjs Framework]
 9: 0x119845fd9 v8::internal::SetupIsolateDelegate::SetupHeap(v8::internal::Heap*) [/Users/jpcote/Desktop/TESTNWJS/nwjs0.39.2.app/Contents/Versions/75.0.3770.100/nwjs Framework.framework/Versions/A/nwjs Framework]
10: 0x11980c87e v8::internal::SetupIsolateDelegate::SetupHeap(v8::internal::Heap*) [/Users/jpcote/Desktop/TESTNWJS/nwjs0.39.2.app/Contents/Versions/75.0.3770.100/nwjs Framework.framework/Versions/A/nwjs Framework]
Received signal 6
 [0x00011a4cc1e9]
 [0x00011a3ed903]
 [0x00011a4cc101]
 [0x7fff6c94eb5d]
 [0x000116f1af87]
 [0x7fff6c8086a6]
 [0x000112c92f1d]
 [0x000112c934b9]
 [0x000118e25ce0]
 [0x000118e25e09]
 [0x000112c310cf]
 [0x000112c34a74]
 [0x00011913fcf6]
 [0x0001192bebb4]
 [0x000119845fd9]
 [0x00011980c87e]
[end of stack trace]
[0709/211007.172465:WARNING:process_memory_mac.cc(93)] mach_vm_read(0x7ffee189f000, 0x2000): (os/kern) invalid address (1)
[0709/211007.295961:WARNING:system_snapshot_mac.cc(42)] sysctlbyname kern.nx: No such file or directory (2)
```

djipco, 2019-07-16:
For the sake of completeness, please note that this issue has been fixed in the Node.js code used by NW.js. This means that, starting with the [nightly build of July 17](https://dl.nwjs.io/live-build/nw39/07-17-2019/945a6ea62/v0.39.3/), this problem is resolved. 

Details are here: https://github.com/nwjs/node/commit/b2c926a38f1f73a199f90560f1e6e08cbfdd1aab

closed 2019-07-16 by djipco

djipco, 2020-03-17:
As of now, there is also another way to get this working. By using `firmata-io` and my new `chrome-apps-serialport` module, you will be able to use Johnny-Five inside NW.js without having to recompile anything. Here's how:

```js
const SerialPort = require("chrome-apps-serialport").SerialPort;
const Firmata = require("firmata-io")(SerialPort);
const five = require("johnny-five");

SerialPort.list().then(ports => {

  const device = ports.find(port => {
    return port.manufacturer && port.manufacturer.startsWith("Arduino")
  });

  const board = new five.Board({
    io: new Firmata(device.path)
  });

  board.on("ready", () => {
    console.log("Johnny-Five is ready!");
    const led = new five.Led(13);
    led.blink(500);
  });

  board.on("close", () => console.log("Closed!"));

});
```
Cheers!