rwaldron/johnny-five #1578: Specifying I2C bus

qwsxzasd1, 2019-05-25:
I'm not sure if this has been mentioned before in the docs or somewhere else, but I have looked for a couple of hours and can't find anything. 
Is it possible to specify which I2C bus? I'm getting a Remote I/O error, and from what I understand, this is because of the Raspberry Pi's hardware bug that prevents clock stretching. After adding a software I2C bus, I got my devices to work using different libraries. I need some of the features that this one offers, but I can't seem to find a way to specify the I2C bus anywhere.

dtex, 2019-08-14:
@nebrius

nebrius, 2019-08-15:
Unfortunately this is not possible in Raspi IO because the IO plugin interface itself doesn't include the concept of multiple busses in all I2C methods. Raspi IO supports multiple busses under the hood, but there's no way to get that data to Raspi IO through Johnny-Five unfortunately.

Specifically, we need to add a `bus` parameter to:
- The three `i2cWrite` methods/variants
- The two `i2cRead` methods/variants
- The two `i2cReadOnce` methods/variants

We also need to make the bus parameter in `i2cConfig` required instead of optional.

fivdi, 2019-08-15:
If I'm not mistaken @qwsxzasd1 would like to tell the io-plugin to use a user specified I2C bus for all attached I2C devices rather than having the io-plugin defaulting to using a particular bus. In other words, the goal is not to use multiple I2C busses at the same time, but rather to tell the io-plugin which one of the multiple I2C busses that are available to use.

In this particular case @qwsxzasd1 would like to tell Raspi IO to use I2C bus 3 for all attached devices rather than using I2C bus 1. I2C bus 3 is the software I2C bus and it supports I2C clock stretching. I2C bus 1 is the hardware I2C bus and it doesn't support I2C clock stretching.

This could be achieved by extending the RaspiIO constructor to support an option that allowed the I2C bus to use to be specified. If the option is not specified Raspi IO can default to using bus 1.

closed 2020-09-13 by stale[bot]