rwaldron/johnny-five #1579: LCD 20x4 cursor is offset

Tim3Game, 2019-05-25:
Hello,
I wanted to create a simple arduino project for displaying things like: PC Stats, Server Stats, maybe weather, time and other stuff. I had my box with an Arduino Uno and an LCD 20x4 with an I2C for a long time and didn't used it, so I decided to create something with the help of NodeJS and Johnny Five. I found the example on the LCD I2C, and after I found the correct controller and port, it worked... but with an offset... So when you set `cursor(0,0)` it should be at the top left (0,0), but on my lcd it shows from the middle (10,2 ("box" 11,3)), and when you set `cursor(2,0)` then it's on the top row, but starts from the middle...

**Code:**
```var five = require("johnny-five");
var board = new five.Board({    
    port: 'COM5',
});

board.on("ready", function() {

  var lcd = new five.LCD({
    controller: "PCF8574T",
    rows: 4,
    cols: 20,
  });

  lcd.useChar("heart");

  lcd.cursor(0, 0).print("hello :heart:");

  lcd.blink();

  lcd.cursor(1, 0).print("Blinking? ");
});
```

**Image of LCD:**
[image]

dtex, 2019-05-26:
> after I found the correct controller

Was this "keep trying until you find one that works" or are you definitely running a PCF8574T?

Tim3Game, 2019-05-27:
Im definitely running a PCF8574T, but in the documentation it looks like you can only choose one of the 3 written, but a bit lower I found mine, so I used that (mine)
[image]
[image]

closed 2020-05-29 by stale[bot]