rwaldron/johnny-five #1580: Can J5 handle reading IR raw data from remote control devices?

communque, 2019-05-28:
Trying to capture codes from IR signals from remote control units for a VCR, TV, etc.
New to this.  Trying to learn what's possible.

Currently testing with the following:
	- ESP8266
	- [IR sensor](https://www.amazon.com/gp/product/B06XYNDRGF/ref=ox_sc_act_title_1?smid=A34K5WF5Z9R33P&psc=1) powered by 3.3VDC from ESP, output connected to ESP's analog input (also testing digital)
	- Code:
```
	var J5 = require("johnny-five");
	var {EtherPortClient}=require("etherport-client");
	var thisBoard, thisSensor, thisTime, priorTime, thisIp;
	thisIp="ip.address.of.ESP";
	
	thisBoard=new J5.Board({
		port: new EtherPortClient({
			host: thisIp,
			port: 3030
		}),
		repl: false
	});

	thisBoard.on("ready", function(){
		thisSensor=new J5.Sensor({
			pin:"A0"
		});
		priorTime=new Date().getTime();
		thisSensor.on("change", function(){
			thisTime=new Date().getTime();
			console.log(thisTime-priorTime, this.value);
			priorTime=thisTime;
		});
	});
```

The approach I'm using seems to be incapable of accurately capturing the signals. The `thisSensor.on("change"` etc. does register that the remotes are transmitting, but not the raw data.  Is it that J5, being JS operates at max speed of 1 ms? while the IR signals are coded at µs?  Or the setup / coding approach, etc?

A few of the J5 git pages [here](https://github.com/rwaldron/johnny-five/issues/434) and [here](https://github.com/rwaldron/johnny-five/wiki/Requested-Features) sort of address the issue.  It seems like there are options that involve SDA & SCL and/or I2C, but so far haven't been able to integrate them.

[This page](https://gist.github.com/rwaldron/affe66562302fa6a5932) looks interesting, but so far haven't breen able get any readings on pins 4/5 -- likely for no knowing how to properly connect the sensor to those pins.

Thanks in advance for advice / explanations

communque, 2019-05-30:
Following up...

Using the code below in the hopes of following the IR sensor on a tighter schedule.
The results are inconsistent,
Each click of a button the remote control device triggers a response, but...
1 - Clicking the same button repeatedly gives different responses each time
2 - Clicking different buttons doesn't give identifiably different responses.
(Perhaps the data stream rate, which I presume is 1ms and ??can't be reduced to nano- or even µsecs??)

Example results (Cleaned up)

Button | Press# | Analog | Dur µSec
-- | -- | -- | --
TV Power Button | 1 | 1024 | 18.84
TV Power Button | 1 | 8 |  
  |   |   |  
TV Power Button | 2 | 1024 | 23.50
TV Power Button | 2 | 6 | 93.52
TV Power Button | 2 | 1024 | 18.16
TV Power Button | 2 | 7 |  
  |   |   |  
TV Power Button | 3 | 1024 | 17.41
TV Power Button | 3 | 8 |  
  |   |   |  
TV Power Button | 6 | 1024 | 23.51
TV Power Button | 6 | 7 |  
  |   |   |  
TV Power Button | 7 | 1024 | 15.37
TV Power Button | 7 | 7 | 170.82
TV Power Button | 7 | 1024 | 20.36
TV Power Button | 7 | 9 |  
  |   |   |  
TV Power Button | 8 | 1024 | 16.88
TV Power Button | 8 | 7 |  

Code used to capture the data...
```
var Fs = require("fs");
var Cp = require("child_process");
var J5 = require("johnny-five");
var {EtherPortClient}=require("etherport-client");
var thisBoard, thisIp;
thisIp="ip.address.of.ESP";

thisBoard=new J5.Board({
	port: new EtherPortClient({
		host: thisIp,
		port: 3030
	}),
	repl: false
});

thisBoard.on("ready", function(){
	var thisSensor, thisTime, thisValue;
	thisSensor=new J5.Sensor({
		emitter: 13,
		pin: "A0",
		freq: 1
	  });
	thisSensor.on("data", function() {
		console.log(new Date().getTime(), this.value);
	});
});

```

dtex, 2019-05-31:
I recommend checking out this issue: https://github.com/rwaldron/johnny-five/issues/434

A "receiver" class that support IR is definitely on our wish list.

dtex, 2019-08-14:
This issue has been added to the wishlist.

closed 2019-08-14 by dtex