rwaldron/johnny-five #1581: Can the SDA/SCL (GPIOs 4/5) be reassigned using Johnny-Five?

communque, 2019-05-31:
Using an ESP8266, and wondering if it's possible to reassign the SDA/SCL pins (default 4/5) to different pins?  e.g. 14/15?

dtex, 2019-05-31:
I don't think you can do that.

communque, 2019-05-31:
Thanks for that.  Here are the pages that led me to believe it might be possible.

- [Arduino forum](https://forum.arduino.cc/index.php?topic=427607.0)
- [StackExchange](https://arduino.stackexchange.com/questions/39210/problems-with-i2c-connection-on-esp8266-12f-which-pins-should-i-use)
- [esp8266.com](https://www.esp8266.com/viewtopic.php?f=32&t=18893)
- [Adafruit](https://learn.adafruit.com/adafruit-feather-huzzah-esp8266/pinouts#i2c-and-spi-pins-2-12)

I haven't tested yet, in part because I'm not 100% sure if I'm understanding any of this correctly, but it seems to suggest editing the StandardFirmata file using the Arduino ID and re-flashing the ESP would do it.

Ideally I was looking for a J5-only solution -- one that might make it possible change it on the fly.

dtex, 2019-05-31:
Ah, there's nothing in the [firmata protocol](https://github.com/firmata/protocol/blob/master/i2c.md) that would allow Johnny-Five to specify I2C pins. It would have to be done in the firmata sketch.

closed 2019-08-14 by dtex