rwaldron/johnny-five #1582: Dimmer control for an LED on an AC circuit using ESP8266

communque, 2019-06-05:
I'm hoping it's possible to control Dimmable AC-powered LEDs using J5 on an ESP8266.

Using the following:

- 	ESP8266 Microcontroller
- 	[RobotDyn AC Dimmer](https://robotdyn.com/ac-light-dimmer-module-1-channel-3-3v-5v-logic-ac-50-60hz-220v-110v.html)

**Question**: First up, Is this even possible using J5 over WiFi with an ESP8266?
I've read there are latency issues that cause flickering.

Assuming it is nonetheless possible...

Here's what I understand so far:
	Connect the Dimmer's PWM pin to a PWM-enabled ESP GPIO
	Connect Dimmer's ZC pin to another ESP GPIO.
	**Question**: Is this GPIO to be set for Digital Read? or Write?

On the J5 side, then what?

dtex, 2019-06-18:
> Is this even possible using J5 over WiFi with an ESP8266?

Yes. There is no official documentation, but check out [this post](https://boneskull.com/how-to-use-an-esp8266-with-johnny-five/), or [this one](https://www.instructables.com/id/ESP8266-Firmata-J5-NodeBot/).

> I've read there are latency issues that cause flickering.

Yes, that's a reasonable concern. The ESP8266 will be running firmata which takes simple commands like "turn pin 1 on". There is nothing like a "fade pin 1 to 0" in firmata so instead Johnny-Five would send many commands in short order, decreasing the value by one each time until it reached zero. Depending on how quickly you are fading, the J5 + firmata.js + wi-fi combo may not be able to keep up.

> Digital Read? or Write?

You would want to read the ZC pin

> Is there an equivalent to attachInterrupt() in J5?

No. and on first blush it looks like the timing is critical enough that you could not monitor ZC with a digital pin and respond in "just Javascript".

>  is there a way to directly call the command from J5?

It would have to be added to firmata first and I think a custom build of firmata that added a fade command (with target and duration) would be the best approach for this.

closed 2019-08-14 by dtex

ScreamZ, 2023-04-03:
@communque Thanks for this quick reply,
I would be pleased to take a look at this.

It's not a problem flickering my use case is for an escape room with some "haunted" ambiance so I'll looking for this kind of junky effect;

I was looking to use a "relay/bridge arduino" that would read a PWM from another Arduino on a specific pin and dispatch the proper dim to be light dimmer AC. This way I could drive the light module dimmer using J5

communque, 2023-04-13:
@ScreamZ Hi Adréas, apologies for the delay.  Got caught up in work and lost this thread.

I ended up not using J5 for testing A/C dimming.  If you can live w/ flickering you're in luck.

The C++ code is brief and relatively straightforward.
Of course you have to get familiar with the terms AND you have to build the MCU code to support the snippet below, but here are the basic commands.  The three methods below are built-in.

```
ledcSetup(PwmChannel, Freq, Res);
ledcAttachPin(Gpio, PwmChannel);
ledcWrite(PwmChannel, DutyCycle);
```
More here: https://randomnerdtutorials.com/esp32-pwm-arduino-ide/
and here: https://espressif-docs.readthedocs-hosted.com/projects/arduino-esp32/en/latest/api/ledc.html

I ended up building a whole system in C++ that can receive cURL calls from terminal or NodeJS child_process.exec / child_process.spawn.  The cURL allow you to run minimal code on the MCU, leaving the bulk of the logic to the JS context.

I know it's a bit brief as far as an explainer, but I hope it's at least somewhat helpful.

Let me know if you have questions