rwaldron/johnny-five #1584: Can't Resolve 'repl'

Mykasoda, 2019-06-08:
Greetings,

I just installed johnny-five via NPM for use on a Raspberry Pi 3b+

After installing I get the following error...

Failed to compile.

./node_modules/johnny-five/lib/repl.js
Module not found: Can't resolve 'repl' in '<root>/node_modules/johnny-five/lib'

When I look in the folder the file repl.js is there. I can't figure out what's going on.

Any advice would be much appreciated

dtex, 2019-06-08:
> Failed to compile

That's strange. Can you tell us more about what you are doing when the error is thrown?  Are you trying to use Webpack or something?

Mykasoda, 2019-06-08:
Ah duh, I bet Webpack is causing the error. Hrm. I guess I'll have to research how to use it in an app that also uses Webpack. Thank you

Mykasoda, 2019-06-08:
ah duh, Yes, using Webpack. Any advice on how to use it in an app that uses Webpack? (it's a ReactJS app). 

In the meantime i'll try to research myself, thanks for the heads up

Mykasoda, 2019-06-08:
Nevermind, googling around shows that it's not possible with a web app. I'll make an API to do it, sorry to clutter issues page!

closed 2019-06-08 by Mykasoda