rwaldron/johnny-five #1585: PWM values for DC Motor go awry when pin assigned to motor is initialised after Servo pin

amolpatravali97, 2019-06-14:
I have a Raspberry Pi zero w running the program and a PCA9685 as the controller to run a DC motor and a Servo motor. When the Servo motor is initialised first, followed by the DC motor, the DC motor rotates at twice the RPM it would if it was initialised first. I checked the voltage coming out of the pins 10 and 9 and the difference would double up for the same speed if the DC motor was initialised second.
Here is the code and the circuit diagram(Fritzing) has been attached.

```javascript
var five = require('johnny-five'),
        keypress = require('keypress'),
        board, motor, servo,
        Raspi = require('raspi-io').RaspiIO;

board = new five.Board({ io: new Raspi() });

board.on('ready', function() {
  servo = new five.Servo({
    pin: 15,
    controller: 'PCA9685',
    startAt: 50,
    range: [0, 105]
  });
  motor = new five.Motor({
    pins: [10, 9],
    controller: 'PCA9685',
    invertPWM: true
  });
  motor.forward(30);
});
```

Here is the code wherein the DC motor works fine, i.e it gives a low RPM at speed 30 as opposed to the previous wherein it was double the RPM for the same speed.
```javascript
var five = require('johnny-five'),
        keypress = require('keypress'),
        board, motor, servo,
        Raspi = require('raspi-io').RaspiIO;

board = new five.Board({ io: new Raspi() });

board.on('ready', function() {
  motor = new five.Motor({
    pins: [10, 9],
    controller: 'PCA9685',
    invertPWM: true
  });
  servo = new five.Servo({
    pin: 15,
    controller: 'PCA9685',
    startAt: 50,
    range: [0, 105]
  });
  motor.forward(30);
});
```
Fritzing:
[image]

dtex, 2019-06-14:
Servos operate in a small section of the possible PWM range but a motor will use 100% of the range. The PCA9685 Expander in Johnny-Five configures a single PWM range for the device rather than a PWM range for each pin. It's not a limitation of the PCA9685, but rather a result of a design decision in Johnny-Five.

I think it's possible to change this, but I wouldn't expect it to happen anytime soon (Unless you're interested in taking it on).

You could always hook either the motor or servo up to GPIO on the Pi.

dtex, 2019-06-18:
I've added a request for this feature to our "Most Wanted" wiki page and referenced back to this issue, so I'm going to close this for now.

closed 2019-06-18 by dtex