rwaldron/johnny-five #1586: Jhonny Five allow us to update firmeware over the air (OTA)?

marquesm91, 2019-06-20:
I love Johnny Five and wonder if Johnny Five has already consider something like OTA.

I don't know if this is out of scope of Jhonny Five project and I don't know either if I can ask it here.

This will be super dope if Jhonny Five could give some native implementation to allow OTA updates!

dtex, 2019-06-20:
What device are you using? Are you wanting to update firmware on a fleet of products in the wild or a personal project?

marquesm91, 2019-06-21:
I will try it on a ESP8266

It is for my final paper and I would like to have a nice alternative to have OTA (FOTA in this case?) coding with javascript in IoT ecossystem.

I reached this [tutorial](http://arduino.esp8266.com/Arduino/versions/2.0.0/doc/ota_updates/ota_updates.html#http-server) and got confused. So, OTA is more like a ESP8266 scope challenge or we can work in something more abstract in Johnny Five?

dtex, 2019-06-21:
With Johnny-Five and the ESP8266 you are running programs on two different systems. On the ESP8266 you are running firmata which is a sketch that just sits there and listens for instructions from the host computer. The host computer is where Johnny-Five lives.

Johnny-Five does not have a way to push firmware to the ESP8266, but there may be other modules that do and you could steal ideas from those (see [nodebots-interchange](https://github.com/nodebots/nodebots-interchange)). 

As for updating code on the host computer, I would probably just "cheat" and use npm. Write the meat of your app and publish it as a module on npm. Then just have a simple boot app that is able to launch your program and "listen" for update commands. When you want to change your app, publish a new version to npm. Your boot app can check npm for a new version and trigger a call to npm update.

Note: I would not use this technique in production.

marquesm91, 2019-06-21:
Thanks for clarifying! I will close this issue because is not a Jhonny Five problem scope

closed 2019-06-21 by marquesm91