rwaldron/johnny-five #1588: mini summit 2019 ?

monteslu, 2019-07-24:
I'll be getting to the conf Aug 11.  Could be there a few hours before the opening party.

Any interest in a get together beforehand to go over last years goals and progress?

@rwaldron, @lyzadanger, @nebrius, @noopkat, @dtex, @ajfisher, @lynnaloo, @sandeepmistry, @HipsterBrown, @reconbot @soundanalogous @julianduque @KatieK2 @ghtomcat @nodebotanist @rockbot

dtex, 2019-07-24:
I'm not getting in until late on the 11th, but I could look at changing my flight.

dtex, 2019-07-24:
For those of y'all that weren't there last year: 
* [Day 1](https://github.com/rwaldron/johnny-five/issues/1463#issuecomment-414544943)
* [Day 2](https://github.com/rwaldron/johnny-five/issues/1463#issuecomment-415898310)

monteslu, 2019-07-24:
brunch on the way out like last time works too.  I'm not in a hurry to leave San Diego to head back to Phoenix :)

ajfisher, 2019-07-27:
I'm flying in on the Sunday but will likely need to do NodeBoats prep as we
have to make up the kits etc ready for Day 2. I am pretty flexi on the way
out as just heading back to LA before going back to AU on Friday night.

reconbot, 2019-07-27:
Late on the Aug 11th (Sunday after 3pm PST) might work for me but Sara has boats and I'll have my kid (which should be fine). We leave at noon on Thursday the 15th so that's tough. Happy to meet during the week. Lets collect a little more input and then start a doodle!

Excited to see everyone! 🙌

lyzadanger, 2019-07-27:
I'm free on Sunday afternoon/evening, so open to various options here...

noopkat, 2019-07-28:
I'm still unsure if I'll be able to make it to JsConf, but I'll attend if I can!

nebrius, 2019-07-28:
I'm flying in Sunday and land at 1pm PDT, getting to the hotel late afternoon-ish, and I'll be there for brunch on Thursday. I'm happy to meet once I get there/before I leave.

reconbot, 2019-08-14:
4pm break tomorrow?

noopkat, 2019-08-14:
I’m down

reconbot, 2019-08-14:
How about lunch? We'll have more time.

(grab lunch and then) 12:10PM at the office/storage room

nebrius, 2019-08-14:
Just a heads up, I’ll probably be a few minutes late. I promised to make an appearance at another lunch thing

dtex, 2019-08-14:
## Johnny-Five Collaborator Summit Takeaways and ToDo's
- Languishing Issues and PR's
  - [x] Add [Stale](https://github.com/apps/stale) to Johnny-Five Repo. Issues that have no activity in the past year will be marked as "Stale". Issues that have been stale for 7 days will be closed. Users can comment while an issue is "Stale" to keep it fresh, and issues can be re-opened once closed. @dtex
  - [x] Tag the existing release of Johnny-Five as v1.0.0. This immediately puts us into "semver mode" and we can start making breaking changes. This will happen Wednesday night, after the workshops. @rwaldron
  - [x] Write a johnny-five.io news post talking about this and our commitment to match support with official node LTS versions. @lyzadanger & @nebrius
- Zombie platforms
  - [ ] Galileo - Reach out to Derek at Sparkfun and ? at Intel and make sure they know they need to lock down on version 0.14.3 in order to ensure compatability for Galileo and Edison (until someone adds support for new Servo methods from c33155e5e8eb6d8a1f594793ec64655aa6175d06) @rwaldron
  - [x] Nino - Mark Repo and package as not maintained/deprecated @rwaldron
- Breaking changes - Our new semver approach and tagging as 1.0.0 puts us in place where we can actually make breaking changes so it will be possible to move forward with these changes individually instead of landing everything before we can tag as 1.0.0. This has been a blocking problem for years and is a huge, positive shift for J5.
  - [ ] Replace all ocurrences of ```freq``` in constructor options with ```period``` and update examples and docs @hipsterbrown
  - [ ] Remove all ocurrences of ```this.emit("read")``` @hipsterbrown
  - [ ] Find all the callbacks and add them to a spreadsheet. Include the args passed to the callbacks and wether the callback can be called more than once. We are working towards following node's standard error first pattern, even if we are just passing null for things that will never error @ajfisher
  - [ ] Update contributing.md to note that any breaking commits must have a short description that is prefixed with ```[Breaking]``` @lyzadanger
  - [ ] Update sensor data event to pass an object like ```{ value: scaledValue, raw: ADCValue }``` where "scaledValue" is [0, 1.0] and "ADCValue" is the raw value which will vary with the ADC bit depth @ajfisher
  - [ ] Update all examples to destructure objects passed to event listeners, grabbing only the values needed @ajfisher (Delegate this task by class maybe?)
- IO plugin feature parity. We clarified the position re: making changes to io-plugins. Rather than a "firmata first" approach where new features were never added to other io-plugins before they were available in firmata.js, we can now add features in io-plugins first. Care must be taken to coordinate with other plug-in maintainers to ensure that we create an API that will work for all. This API spec will be maintained in rwaldron/io-plugins.
  - [ ] Revisit rwaldron/io-plugins to make sure it is current and complete and reflects the format we want to use for the "Standard" @rwaldron
  - [x] Create abstract-io @nebrius
  - [ ] Open issue on each io-plugin repo informing maintainers of our clarified approach to new features and the new/improved io-plugins repo @dtex
- SPI Support in io-plugins
  - [ ] Add SPI support to raspi-io @nebrius
  - [ ] Add SPI support to imp-io @dtex
  - [ ] Add SPI support to tessel-io @rwaldron
  - [ ] Add SPI support to ble-io @monteslu
  - [ ] Add SPI support to firmata.js @rwaldron
-  Board scriptable behavior
   - [ ] Draft a scriptable package protocol @ajfisher
   - [ ] Test protocol on imp-io @dtex
- IMU/Multi organization
  - [ ] Explore breaking individual multis out to component plugins
- PWM support for platforms with > 8-bit resolution
  - [x] Just effing do it @dtex
- ES6 J5 status / supported node versions - We now officially support all Active, Maintenance and Current [LTS releases](https://github.com/nodejs/Release#release-schedule) and we drop official support when they hit EOL.
- Update examples - Stop using ```this```, use arrow functions where appropriate, spread operators, ```const``` on requires, and appropriate variable scoping (```let```)
  - [ ] Update ```sensor``` class examples  @ajfisher
  - [ ] nit new ```sensor``` class examples @rwaldron
  - [ ] Set plan for updating all the other examples (assign by class) @dtex
- Component plugin super classaa
  - [ ] Draft specification @noopkat
- Alternative tranports within serialport
  - [ ] Add bindings for WebUSB @monteslu @reconbot
  - [ ] Based on this add bindings for other transports TBD
- Handle disconnects gracefully
  - [ ] Propose heartbeat feature for firmata that allows users to define a heartbeat interval and a target state for disconnected devices (The OFS) @dtex
  - [ ] Ping @sandeepmistry to see if there is a way for the board to know if a serial connection has been lost. See ```cdc.cpp``` line state which is currently marked as a to-do. (Note that this is not a panacea as off-board devices will still be connected to serial even if their connection to the host has been lost).

dtex, 2019-08-15:
We are dropping some of the to-do's above:

Things we are not going to do:

Reach out to Derek at Sparkfun and ? at Intel and make sure they know they need to lock down on version 0.14.3 in order to ensure compatability for Galileo and Edison.

* Replace all occurrences of freq in constructor options with period and update examples and docs
* Board scriptable behavior
  Draft a scriptable package protocol @ajfisher
  Test protocol on imp-io @dtex

* Board scriptable behavior
  Draft a scriptable package protocol @ajfisher
  Test protocol on imp-io @dtex

dtex, 2019-08-15:
## Johnny-Five Collaborator Summit Takeaways and ToDo's from 2019
- Breaking changes - Our new semver approach and tagging as 1.0.0 puts us in place where we can actually make breaking changes so it will be possible to move forward with these changes individually instead of landing everything before we can tag as 1.0.0. This has been a blocking problem for years and is a huge, positive shift for J5.
  - [ ] Remove all ocurrences of ```this.emit("read")``` @hipsterbrown
  - [ ] Find all the callbacks and add them to a spreadsheet. Include the args passed to the callbacks and wether the callback can be called more than once. We are working towards following node's standard error first pattern, even if we are just passing null for things that will never error @ajfisher
  - [ ] Update contributing.md to note that any breaking commits must have a short description that is prefixed with ```[Breaking]``` @lyzadanger
  - [ ] Update sensor data event to pass an object like ```{ value: scaledValue, raw: ADCValue }``` where "scaledValue" is [0, 1.0] and "ADCValue" is the raw value which will vary with the ADC bit depth @ajfisher
  - [ ] Update all examples to destructure objects passed to event listeners, grabbing only the values needed @ajfisher (Delegate this task by class maybe?)
- IO plugin feature parity. We clarified the position re: making changes to io-plugins. Rather than a "firmata first" approach where new features were never added to other io-plugins before they were available in firmata.js, we can now add features in io-plugins first. Care must be taken to coordinate with other plug-in maintainers to ensure that we create an API that will work for all. This API spec will be maintained in rwaldron/io-plugins.
  - [ ] Revisit rwaldron/io-plugins to make sure it is current and complete and reflects the format we want to use for the "Standard" @rwaldron
  - [ ] Open issue on each io-plugin repo informing maintainers of our clarified approach to new features and the new/improved io-plugins repo @dtex
- SPI Support in io-plugins
  - [ ] Add SPI support to raspi-io @nebrius
  - [ ] Add SPI support to imp-io @dtex
  - [ ] Add SPI support to tessel-io @rwaldron
  - [ ] Add SPI support to ble-io @monteslu
  - [ ] Add SPI support to firmata.js @rwaldron
- IMU/Multi organization
  - [ ] Explore breaking individual multis out to component plugins
- ES6 J5 status / supported node versions - We now officially support all Active, Maintenance and Current [LTS releases](https://github.com/nodejs/Release#release-schedule) and we drop official support when they hit EOL.
- Update examples - Stop using ```this```, use arrow functions where appropriate, spread operators, ```const``` on requires, and appropriate variable scoping (```let```)
  - [ ] Update ```sensor``` class examples  @ajfisher
  - [ ] nit new ```sensor``` class examples @rwaldron
  - [ ] Set plan for updating all the other examples (assign by class) @dtex
- Component plugin super classaa
  - [ ] Draft specification @noopkat
- Alternative tranports within serialport
  - [ ] Add bindings for WebUSB @monteslu @reconbot
  - [ ] Based on this add bindings for other transports TBD
- Handle disconnects gracefully
  - [ ] Propose heartbeat feature for firmata that allows users to define a heartbeat interval and a target state for disconnected devices (The OFS) @dtex
  - [ ] Ping @sandeepmistry to see if there is a way for the board to know if a serial connection has been lost. See ```cdc.cpp``` line state which is currently marked as a to-do. (Note that this is not a panacea as off-board devices will still be connected to serial even if their connection to the host has been lost).

nebrius, 2019-08-15:
I've started preliminary work on the new Raspi IO CLI at https://github.com/nebrius/raspi-io-config. I've got some really cool ideas!

reconbot, 2019-08-16:
I got the github action deploy process for the J5 website working  🚀

PR here https://github.com/rwaldron/johnny-five/pull/1597

It gets updated if *anyone* updates the wiki or if you push or merge to master

Logs of the deploy can be found in the ["actions" tab](https://github.com/reconbot/johnny-five/commit/3d9d89b8e75e1eb5f1ce28f0663295f5602e2dbc/checks)

The action's code can be found here https://github.com/reconbot/johnny-five/blob/master/.github/workflows/build-website.yml

HipsterBrown, 2019-08-17:
I haven't found any current instances of `emit("read")` in the codebase. It looks like Rick replaced them all with `emit("data")` a while ago -> https://github.com/rwaldron/johnny-five/commit/2b4230dd9da98c459de7be2b312f5494b069a03c

The closest example of `emit("read")` is the `digital-read` event emitted by the Expander class. https://github.com/rwaldron/johnny-five/blob/5f4512db35f488ff065c55a8a1396c8eb8c2fd6e/lib/expander.js#L207

reconbot, 2020-08-12:
lol stalebot

reconbot, 2020-08-12:
I miss all of you and wish we could have jsconf this year but look forward to the future ❤️

closed 2020-08-12 by reconbot