rwaldron/johnny-five #1590: Bad logic in compass.js object reference

nkolban, 2019-08-01:
When trying to use the compass object on a Raspberry Pi, I am getting the following error:

```
/home/pi/rcboat/node_modules/j5-io/dist/index.js:481
            throw new Error('Invalid arguments');
            ^

Error: Invalid arguments
    at J5IO.i2cWrite (/home/pi/rcboat/node_modules/j5-io/dist/index.js:481:19)
    at Compass.value (/home/pi/rcboat/node_modules/johnny-five/lib/compass.js:50:17)
    at new Compass (/home/pi/rcboat/node_modules/johnny-five/lib/compass.js:532:10)
    at Board.board.on (/home/pi/rcboat/run.js:33:16)
    at emitNone (events.js:111:20)
    at Board.emit (events.js:208:7)
    at _combinedTickCallback (internal/process/next_tick.js:132:7)
    at process._tickDomainCallback (internal/process/next_tick.js:219:9)
    at Function.Module.runMain (module.js:696:11)
    at startup (bootstrap_node.js:204:16)
Makefile:2: recipe for target 'run' failed
make: *** [run] Error 1

```

Digging into the source code at compass.js I am looking at the HMC5883L object.  It is defined as follows:

```
  HMC5883L: {
    REGISTER: {
      value: {
        // Page 11
        // Table 2: Register List
        //
        // Configuration Register A
        CRA: 0x00,
        // Configuration Register B
        // This may change, depending on gauss
        CRB: 0x01,
        // Mode Register
        MODE: 0x02,
        // Data Output X MSB Register
        READ: 0x03,
      }
    },
```

I then see it used as follows:

```
this.io.i2cWrite(address, this.REGISTER.CRA, 0x70);
```

and it is here that I am having a puzzle.

How are we access the CRA field?  It looks like the accessor should be:

REGISTER.value.CRA as opposed to REGISTER,CRA?

dtex, 2019-08-01:
Hi @nkolban ,

The controller object is applied to the compass instance in Board.Controller with a call to [Object.defineProperities](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/Object/defineProperties). Check it out, it's a cool way to "decorate" an object.

dtex, 2019-08-01:
It seems like this was just a question about code so I'm going to go ahead and close it.

If this was just the beginning of a larger issue, feel free to respond and re-open.

closed 2019-08-01 by dtex