rwaldron/johnny-five #1591: PCA9685 and HT16K33 Not Working Together

becauseinterwebs, 2019-08-02:
I have the [PCA9685 16 channel I2C interface board](https://www.adafruit.com/product/815) and the [8x8 led matrix](https://www.adafruit.com/product/870) (using HT16K33) from Adafruit.  I cannot get both of them working together.

The PCA9685 is controlling an RGB Led and I have it switching colors every second.  Whenever I do anything with the 8x8 (HT16K33), the LED just freezes at the current color and does nothing.  The 8x8 led works fine in all cases.

The address of the PCA9685 is 0x40 and the LED matrix is at 0x70.  I'm running this on a Raspberry Pi Zero W using Raspi-IO.  

I've tried connecting the 8x8 to the pass-thru headers of the PCA9685 and also running the SDA/SCL (as well as VCC and GND) into a bread board and then hooking the 8x8 and PCA up to the breadboard.  Same results.

It seems the HT16K33 is blocking the PCA9685 somehow?

Here is the code:

```
const Raspi = require('raspi-io').RaspiIO;
const five = require('johnny-five');
const board = new five.Board({
    io: new Raspi()
});

// Heart shape for 8x8
var heart = [
    "01100110",
    "10011001",
    "10000001",
    "10000001",
    "01000010",
    "00100100",
    "00011000",
    "00000000"
  ];

 var matrix = new five.Led.Matrix({
    addresses: [0x70],
    controller: "HT16K33",
    rotation: 4,
  });

// index for color switching
let index = 0;
let rainbow = ["FF0000", "FF7F00", "FFFF00", "00FF00", "0000FF", "4B0082", "8F00FF"];

// setup the RGB connected to the pins on the PCA9685
let rgb = new five.Led.RGB({
    pins: {
        red: 0,
        green: 1,
        blue:  2
    },
   addresses :  [0x40], // with or without this line doesn't seem to matter
    controller : "PCA9685"
});

// set color of RGB LED and strobe at random interval
const setColor = () => {
    rgb.color(rainbow[index]);
    rgb.strobe(Math.floor(Math.random() * 2000));  
    index++;
    if (index === rainbow.length) {
        index = 0;
    }
    setTimeout(setColor, 2000);
}

const startMatrix = () => {
  matrix.clear(); // led freezes here
  matrix.draw(heart);
}

board.on('ready', () => {
    setColor();
    // using a 10 second delay during which time the Led blinks just fine
   setTimeout(startMatrix, 10000); 
});

board.on('exit', () => {
    // When the board disconnects, turn the LEDs off.
    rgb.stop().off();
    matrix.clear().off();
    console.log('[johnny-five] Board closing - bye bye!');
});
```

dtex, 2019-08-02:
It seems plausible that there could be a bug that just hasn't popped up before. I don't think many people have tried using multiple I2C devices yet. 

I'll test out the Johnny-Five side tonight. We may need to enlist @nebrius if it works in Johnny-Five w/o raspi-io.

becauseinterwebs, 2019-08-02:
Great thank you!  I am going to start digging into the code tonight and try it on an arduino as well.

dtex, 2019-08-04:
Sorry it took a while. I tested with what I have and I see that the I2C writes to the PCA9685 do not stop after the matrix is initialized so that eliminates Johnny-Five as the cause of that particular problem.

I do think I see another, altogether different bug. While the strobing works, are you seeing the colors you expect?

becauseinterwebs, 2019-08-04:
No worries and thanks again for the help! 

While strobing it looks like the colors are correct.

becauseinterwebs, 2019-08-08:
Ok I tested it using an Arduino Mega and connecting the PCA9685 to it and an rgb led connected to that. Works fine.  Chained the HT16K33 to the PCA9685 and getting the same results. Here is a vid of what's happening:

https://youtu.be/4cxK9wLCnhY

dtex, 2019-08-08:
Wow, thanks for the video! It looks like the TX/RX LEDs on the UNO are still blinking on the correct interval, even after the RGB LED stops responding, so I assume the writes are still happening. I wonder if the I2C address of the PCA9685 is getting stepped on in J5 when we initialize a second device. I admit I didn't look at the address when I looked at the I2C messages.

I'll check that out tonight.

becauseinterwebs, 2019-08-08:
You are correct about the led calls still being made. After the matrix is cleared and writes out, the blink method of the RGB is still be called. I have traced all the way into the j5-io library so far and the i2cwrite method is still being called (I just have the led blinking red at this point):

i2cWrite(address, registerOrInBytes, inBytes)

output:
i2cwrite 64 [ 6, 0, 0, 4096, 16 ] undefined
i2cwrite 64 [ 10, 0, 0, 4096, 16 ] undefined
i2cwrite 64 [ 14, 0, 0, 4096, 16 ] undefined

i2cwrite 64 [ 6, 4096, 16, 0, 0 ] undefined
i2cwrite 64 [ 10, 0, 0, 4096, 16 ] undefined
i2cwrite 64 [ 14, 0, 0, 4096, 16 ] undefined

And I'll admit I didn't check the address, either :) Was going to try and dig in a little further tonight.

dtex, 2019-08-08:
Now that I've seen the wiring, I wonder if it could be a power issue. You're not using an external power supply on the PCA9685 so the Uno, the PCA9685, the RGB LED, and the LED Matrix are all drawing power from your USB port which is capped at 500ma. I couldn't find specs on the matrix power draw, but if it's about 20ma per LED and you're turning on 16 of them, that 320ma plus another 20 or so for the RGB LED and I don't know how much for the PCA9685 and the Uno could be overloading your USB port. Maybe the PCA 9685 is "smart" and shuts itself off in low power situations.

While I couldn't find anything on Adafruit's site about the current draw for the matrix, I did see in [this instructable](https://www.instructables.com/id/Arduino-8x8-LED-Matrix/) that an external 5V, 1A power supply is recommended. When you add an external power supply, make sure you add resistors on your LED so nothing bad happens to it.

becauseinterwebs, 2019-08-09:
ah that's a good point. The pca has a VCC terminal, let me try hooking up a power supply to that and running the matrix and led from it.

becauseinterwebs, 2019-08-09:
Well...connected a 5V 1A power supply to the PCA9685 and am powering the led matrix from that, but same results.  I also connected the positive lead directly to the matrix and still the same.

dtex, 2019-08-09:
Bummer. Okay, I've ordered my own so I can replicate the problem. It should be here late next week.

becauseinterwebs, 2019-08-09:
No worries and I'll keep digging. Thank you for the help!

becauseinterwebs, 2019-08-16:
@dtex I think I'm getting closer, and I think it's actually in the firmata-io lib.

When just the LED on the PCA9685 is blinking the i2cRequest method in firmata-io/firmata.js around line 2531 has an active board state of:

`{ '64': { stopTX: true }, delay: 0 }`

(where 64 is the address of the PCA9685)

When the writes to the HT16K33 kick in, the active board state looks like this:

`{ '64': { stopTX: true }, '112': { stopTX: true }, delay: 0 }`

And it looks like only the second board is being written to.  I'm still trying to track this down, but just wanted to update you in case you got started.

dtex, 2019-08-19:
Just wanted to let you know everything showed up and I got it all soldered together. I am seeing the same behavior. Diving in...

dtex, 2019-08-19:
* The bytes being sent over i2c for the RGB LED look the same both before and after the matrix is initialized
* If I initialize the matrix, draw a character, and then start blinking the LED, it works until I try to update the matrix with a new character
* The matrix seems to always work fine regardless of initialization order (I can change characters even after the PCA9685 freezes)

I need to get some sleep, but tomorrow evening I will try changing things on the J5 side so that i2c_config is sent only once. I'm also going to try a different i2c device (instead of the PCA9685).

becauseinterwebs, 2019-08-19:
Great thanks for the update.  I think I got about as far as you and have not had time to continue.  Hoping to get back to it in the next couple of days!

dtex, 2019-08-20:
I was able to use another i2c device in concert with the matrix and both worked fine, but I could not use the other device with the PCA9685 (same result).

I think my next test will be to create an Arduino C sketch that does essentially the same thing as your program and see if that works.

I did find some folks who were talking about PCA9685 freezes and attributing it noise on SDA/SCL that could be fixed with pull-up resistors. The Arduino, of course, has pull-up resistors that are used for i2c, but some folks said those resistors were not adequate.

becauseinterwebs, 2019-08-20:
Ah...that makes a lot of sense.  This is my first time working with PCA9685, and after reading your response and doing some research I found several sources that talked about how the signal gets dirtier the longer your chain gets (event though my chain is "2" at this point...)  I found some hopefully usable info here:

http://dsscircuits.com/articles/effects-of-varying-i2c-pull-up-resistors 

https://community.particle.io/t/has-anyone-used-the-adafruit-pca9685/15539/7

I have another PCA9685 that I will try instead of the HT16K33 so see if that makes a difference, and will also try the pull-up resistors on the SDA and SCL lines.

becauseinterwebs, 2019-08-23:
Well...still no joy :(  I hooked up various resistors from 220 Ohm to 4.7k to pull-up from 5V for SDA and SCL and still same results.  Also tried on 3.3V just for grins...  Will keep experimenting.

becauseinterwebs, 2019-08-23:
Ok...little bit of progress.  Hooked up another PCA and got interesting results :)

https://youtu.be/xTl-2A9je6o

dtex, 2019-08-25:
I put together a little test sketch and it works when running on the arduino

https://github.com/rwaldron/johnny-five/issues/1591

So the problem could be in firmata, firmata.js or Johnny-Five but you saw the same results on the Pi with raspi-io correct?

dtex, 2019-08-25:
I think I've got it!

The PCA9685 is being configured to listen to all commands regardless of address. Try this and let me know if it works...

In ```/node_modules/johnny-five/lib/expander.js``` change line 633 from:
````js
this.io.i2cWriteReg(this.address, this.REGISTER.MODE1, 0xa1);
````
to:
````js
this.io.i2cWriteReg(this.address, this.REGISTER.MODE1, 0xa0);
````

That should fix both Arduino and Raspberry Pi. If that works, I'll get a PR in.

dtex, 2019-08-25:
I ported this bug from Adafruit's library to J5 back in 2014. @ladyada fixed it last year but I didn't get the memo 🤷‍♂ 

https://github.com/adafruit/Adafruit-PWM-Servo-Driver-Library/commit/9f8e1dd41fc8d59a6776142274613b0d97da37a7#diff-5d1e3a5213c0ef6dedb2baf5cc323727L106-R110

becauseinterwebs, 2019-08-26:
Boom!  Yes!  That did it!  Thank you!  I tested on both the Rpi and arduino and they both are working great!
Thank you so much for finding that!

dtex, 2019-08-26:
Awesome! Thanks for your patience and thank you for discovering and reporting this bug. The videos were super helpful. 

I'll get a PR in this morning.