rwaldron/johnny-five #1592: With IIC/I2C 2004 20X4 Character LCD Module Display Blue Green ArduinoMega 2560

FredrikHj, 2019-08-06:
Hi all

I have problem to get my lcd display to work together with: [http://johnny-five.io/api/](http://johnny-five.io/api/) API

The display is:  https://www.ebay.com/itm/With-Without-IIC-I2C-2004-20X4-Character-LCD-Module-Display-Blue-Green-Arduino/263128525818?

The code:
`
boardVs1_2.on("ready", function () {
  let lcd = new jFive.LCD({ 
    controller: "PCF8574AT",
    pins: ['5' ,'6'],
    backlight: 3,
    rows: 4,
    cols: 20,
  });
  //lcd.on();
  //lcd.cursor(0,0);
  lcd.print("Hi");
});
`

If there someone who can help me I will send more info if nedeed :)

Best regards Fredrik Hj

dtex, 2019-08-06:
This is probably the best place to ask. Share the rest of your code and tell us what's happening.

FredrikHj, 2019-08-07:
I got the lcd working for 10 minutes ago :)

The issue now is that I have a underline representing the col. How can I remove it or hide the line?

See the image at: https://1drv.ms/u/s!Ah3ROr-SBUwzo4xDX-szM6ECqq6JLQ

The line is at poss... 3,0!

Is that possible with code remove the background the dark boxes?

Best regards Fredrik

dtex, 2019-08-07:
I don't know a lot about LCD's but I found this in an earlier issue:

https://github.com/rwaldron/johnny-five/issues/1467#issuecomment-475881589

rwaldron, 2019-08-12:
@fivdi?

fivdi, 2019-08-22:
The underline is most likely the cursor. It can be explicitly turned off by invoking `lcd.noCursor()`.